lancaster-university/microbit-dal #218: Tone playback in TD and blocks has clicks

whaleygeek, 2016-10-03:
from: Kevin Spur
ticket: 7
problem: micro:bit sound output
sitever: 0.0.2250.2635

what I am finding is that the tone (using the Block tone command) coming from the micro:bit isn't stable and sounds a bit 'ruff' (I checked this by outputting the same tone from one of our Picaxe boards and this was stable). On my scope I am seeing a frequency variation of about 3% to 4% when the tone is being output. This is much more obvious when the audio signal is amplified.

reproducible.
in blocks: music->play tone(C)
get ticks roughly twice per second.
Same in TD.
rate of ticking changes depending on the tone chosen.

Checked on scope, see artefacts at 20ms (the frame time) and artifacts at roughly 400ms when playing middle C, which turns out to be 1 short low cycle on that period. Duty cycle varies between 50% and 55.5% as a result.

Tried the same in PXT with default settings, the tone is very clean with no ticking. Tried at different frequencies, still very clean and stable

cpseager, 2016-10-03:
@bluetooth-mdw and myself also heard this, we concluded it's a tick from the bluetooth soft device we couldn't do much about. When the sd is not included in the build the tick goes away.

bluetooth-mdw, 2016-10-03:
Not so much "from the bluetooth soft device" (I'm being pedantic now) but a consequence of interrupt priorities and the Bluetooth stack needing a high priority, sometimes to the detriment of other threads. It sounds like this may not be the case though (or at least it's not the only source of the problem) given @whaleygeek 's description of the issue here.

finneyj, 2016-10-03:
Yes, I've observed this too. TD and PXT use the same PWM driver, but TD by default has SoftDevice enabled and PXT has it disabled. As @bluetooth-mdw says, SoftDevice uses the highest possible  interrupt priority for Bluetooth protocol processing. It's also closed source and non-configurable, so nothing we can do on that side (although I'll have a word with our friends at Nordic to see if they've sen this before). 

An improved PWM driver that is tolerant of delayed interrupts has been on my mental radar for a long time, so I'd be very happy to see an improvement here. @jamesadevine I remember you drafted up a new PWM driver with one of the nordic guys when we were down at ARM once that seemed promising, but never quite worked right. Do you still have it tucked way somewhere that you could share?

jamesadevine, 2016-10-03:
https://github.com/lancaster-university/mbed-classic/tree/pwmout-patch

Is where the WIP patch resides.

The idea was to remove the reliance on interrupts through using GPIOTE instead. Alas I didn't get very far, I almost had something but the signals still had the jitter. I was working with a guy at Nordic to come up with this patch, but we lost touch a bit.

whaleygeek, 2016-10-13:
Hi Guys, I was wondering if there is any update on this ticket at all?

finneyj, 2016-10-15:
@whaleygeek I've had a look at the branch @jamesadevine shared above. As noted, it is pretty broken at the moment. :-)

The principle looks pretty good though - essentially removing the ISR in favour of a second GPIOTE/PPI channel such that both the HIGH and LOW transitions of the PWM are pure hardware controlled. This should mitigate any possible effects caused by SoftDevice's high priority interrupts. I'm not sure why it's using two timers though - looks to me like one would suffice.

I've had a play with a simplified derivative of this code, and have three PWM outputs running concurrently at 25%, 50% and 75% duty cycle, so I *think* the approach will work for us. There are some edge cases that still need working out (as everything is now hardware controlled, there are some interesting potential race conditions that need to be controlled). Leave this one with me.

finneyj, 2016-10-18:
@whaleygeek @jamesadevine 

Quick update here. Spent a little time on this last night, and have built up some sample code based on the principles outlined in @jamesadevine's branch, and added some optimizations. Generally, it's looking good. 

Here's a list of the optimizations I made:

- Only uses a single hardware timer (TIMER2 - same as the current mbed one).
- Hardware only implementation for _steady state_ PWM (so should be free from  glitches from SoftDevice interrupts for steady state waveforms such as sound, servo etc).
- Dynamic  timer prescaler calculation to trade off precision against range, so the PWM can hit lower frequencies or higher precision, depending on the PWM freq. chosen. (Old driver could only go down to about 4Hz - I thought a bit less might have uses in education).
- inbuilt round-robin of PWM channels when the user allocates too many PWM channels. This should allow us to get rid of the nasty DynamicPwm wrapper class and get back to the mbed API.
- Updates to PWM duty cycle and frequency now occur inside a one-shot interrupt that's called at start of the next period after a change is made. This means we _can_ still glitch under SoftDevice, but it won't happen anywhere near as often for most applications. SD would need to overlap its interrupt with the start of the first PWM cycle after a change to the PWM duty cycle or period - In this case the start of the next pulse will be delayed until SD's high priority interrupt completes (which I think is about as good as we can get).

I'm pretty much out of time to take this further forward now though... @jamesadevine - can you take my code tomorrow and work it up to fit an mbed driver shaped hole, pop it into a branch and give it some more testing?

@bluetooth-mdw @cpseager Do you think you could maybe do a quick test of this branch against your bluetooth sound apps when James is done?

cpseager, 2016-10-18:
I'm quite happy to test with Scratch when ready, the old ticks are quite audible so any improvement should be instantly recognisable.

bluetooth-mdw, 2016-10-18:
Likewise, happy to help. I'm away at present but have some time next week. That said I suspect I'd only be duplicating what @cpseager would do so if I see he's dealt with this I'll probably leave it.

bluetooth-mdw, 2016-10-24:
I have some time to help @CPSeager with this today. Can I check which branch the optimisations made by @finneyj are in please? The one shared by @jamesadevine above seems to have not been committed to since February. Cheers.

finneyj, 2016-10-24:
Thanks @bluetooth-mdw latest version is here:
https://github.com/lancaster-university/mbed-classic/tree/pwm-stable

A few issue remaining around long periods (over 1s apprarently), but good enough for a funcitonality test

bluetooth-mdw, 2016-10-24:
Seems hugely improved. Video showing the full extent of my testing:

https://drive.google.com/open?id=0B9Z54U0dfwbFWlhEZXk2aVR3cVU

Hex file for @cpseager and others (no pairing required) to test with:

https://drive.google.com/open?id=0B9Z54U0dfwbFVVRpWHRnMzdGMUk

finneyj, 2016-10-24:
Great - thanks @bluetooth-mdw!

@cpseager, if this also seems ok for your application, we can look at merging this into master.

cpseager, 2016-10-24:
Just tested with Scratch (Win7/BLED112 dongle) and the sounds are all much better thanks, the tick has gone.

With this hex I did notice the push switches are much more sensitive to bounce, for instance when I do this:

[image]

the 'h' of hello can re start 2 or 3 times with the briefest of button A taps, when it didn't do this before. However I suspect that may be more to do with #186 and hence require an extra debounce time to be added into the Scratch plugin?

finneyj, 2016-10-24:
Cool, thanks @cpseager, @bluetooth-mdw  - great stuff. 

There haven't been any changes to the debounce logic, so seeing changes there is a little bit weird. I think you may be right about #186, as i *think* the BLE LEDService scrolls asynchronously. An ideal solution here might be to us the EventService to listne for the button click events from the runtime... these are already debounced, and you'll also get A+B events, and optionally touch events on pins P0...P2. 

p.s. I've fixed the only bug we found in the new PWM code just now, so this patch should get merged in with the next release.

jamesadevine, 2016-10-25:
Will there be inter-op issues with dynamic pwm?

James.

https://github.com/lancaster-university/mbed-classic/tree/pwm-stable

finneyj, 2016-10-25:
@jamesadevine 

Hi james - good question. Yes, i suspect here will be a feature interaction, as DynamicPwm assumes a certain implementation (it was always a bit of a hack really wasn't it?). I think this patch renders it unnecessary doesn't it? So we could replace DynamicPwm with PwmOut?

whaleygeek, 2016-11-08:
Hello! Does anyone have an update on this at all? Looks like a fix is possible, was wondering if it was tagged against any specific future release? Thanks!

whaleygeek, 2016-11-20:
Hello! Any updates on this issue yet? We have a customer waiting for a response/resolution. Thanks!

whaleygeek, 2017-01-03:
Hi, any updates on this issue yet? @jaustin @finneyj Thanks!

jaustin, 2017-01-23:
@finneyj I'd like this to be in the next release of the DAL, please (with RC ideally aligning with PXT RC) - What needs to be done for that? Or is it already released? Would be great to close it out or at least have a status update. Looks perhaps like @jamesadevine the ball is with you to respond about PwmOut being an adequate and non-interacting replacement to DynamicPwm?

jamesadevine, 2017-01-24:
@jaustin It's already done: https://github.com/lancaster-university/mbed-classic/commits/pwm-stable

Just needs to be put in the next release, and DynamicPWM removed.

jaustin, 2017-01-24:
Great news! Thanks all - do we have a set of PR's tagged for 'next release'
so we can track what needs to be merged/tested/deployed?

DavidWhaleMEF, 2017-02-13:
Any ETA on this one yet guys? Thanks!

To put you under pressure, our oldest open ticket number in our support system (ticket 7 and we are on tickets over 1000 now) is tagged against this github ticket :)

jamesadevine, 2017-02-13:
@DavidWhaleMEF Hey David, forgot to ping you. This needs to be tested and verified that the patch is correct, perhaps by the person who first filed the issue? Of course we've tested it, but it would be great to get an independent third party to test.

DavidWhaleMEF, 2017-02-13:
Great thanks @jamesadevine I will link you up with customer - where do they download the fix from, if you paste a URL here they can get a .hex file with the fix in it, I can pass it over directly. Thanks!

jamesadevine, 2017-02-13:
@DavidWhaleMEF I'm happy to share a hex, but for an independently verified test it would be great if the customer could do an offline build from the PR here: https://github.com/lancaster-university/microbit-dal/pull/264

DavidWhaleMEF, 2017-02-13:
Thanks, but sorry, I don't believe the customer has the skills or the time to do that :(

finneyj, 2017-02-13:
It'll be in the next tagged release I'm sure.

DaveAtKitronik, 2017-02-23:
@whaleygeek @jamesadevine  I have built and tested PR https://github.com/lancaster-university/microbit-dal/pull/264 and left comments on it.

jamesadevine, 2017-03-01:
@whaleygeek In the latest release, `v2.0.0-rc8`, please reopen if it's still visible in blocks after the PXT have adopted the latest release.

closed 2017-03-01 by jamesadevine