lancaster-university/microbit-dal #220: MICROBIT_ID_NOTIFY_ONE can 'lose' event and leave fibers permanently blocked

drmpf, 2016-10-05:
The following test main illustrates the problem
```
#include "MicroBit.h"
MicroBit uBit;
void waitAny() {
    fiber_wait_for_event(MICROBIT_ID_NOTIFY, MICROBIT_EVT_ANY);
    uBit.display.printCharAsync('x');
}
void displayA() {
    uBit.display.print("ABC");
}
void displayB() {
    uBit.display.print("123");
}
void displayC() {
    uBit.display.print("XYZ");
}
int main() {
    uBit.init();
    uBit.display.printCharAsync('S');
    create_fiber(&displayA);
    create_fiber(&waitAny); // this fiber consumes event meant for displayB and waiting fibers just hang forever
    create_fiber(&displayB);
    create_fiber(&displayC);
    release_fiber(); // finished with setup, release the fibers!!
}
```

Fibers displayB, displayC remain blocked forever, waiting for a wake event up that never comes.

The suggested fix is to change MicroBitFiber.scheduler_event() method's handling of MICROBIT_ID_NOTIFY_ONE
from
`   if ((evt.source == MICROBIT_ID_NOTIFY_ONE && id == MICROBIT_ID_NOTIFY) && (value == MICROBIT_EVT_ANY || value == evt.value))`
to
`   if ((evt.source == MICROBIT_ID_NOTIFY_ONE && id == MICROBIT_ID_NOTIFY) && (value == evt.value))`

As illustrated by the display and serial classes, MICROBIT_ID_NOTIFY_ONE is used to control multi-fiber access to a single resource.

The other use for 
`fiber_wait_for_event(MICROBIT_ID_NOTIFY, xxx)`
is to pause a number fibers waiting on a global trigger (digital input, button, etc)

Without the suggested code change, users can easily break a single resource access control system by naively calling 
`fiber_wait_for_event(MICROBIT_ID_NOTIFY, MICROBIT_EVT_ANY);`
to have a fiber wait for anyone of a number of global triggers, to print the trigger received for example.

Also 
`fiber_wait_for_event(MICROBIT_ID_NOTIFY_ONE, someEventValue);`
is an obvious use if the user wanted to only one fiber to wake up,  but with the current code sending
`MicroBitEvent(MICROBIT_ID_NOTIFY_ONE, someEventValue);`
will wake up **ALL** the fibers waiting on (MICROBIT_ID_NOTIFY_ONE, someEventValue)
This counter intuitive.
The suggested fix for this is to change the above code in MicroBitFiber.scheduler_event() to
 ```
  if ((evt.source == MICROBIT_ID_NOTIFY_ONE
        && (id == MICROBIT_ID_NOTIFY || id == MICROBIT_ID_NOTIFY_ONE)) 
         && (value == evt.value))
```
So that MICROBIT_ID_NOTIFY_ONE becomes a valid eventId and does what the user expects.
Note: also need to change the 
`//Normal case`
 to exclude handling any (evt.source == MICROBIT_ID_NOTIFY_ONE)
i.e. MICROBIT_ID_NOTIFY_ONE only ever wakes one or none fibers.

finneyj, 2016-10-19:
Hi @drmpf 

Thanks for the analysis here! Good observation about combination of NOTIFY and MICROBIT_EVT_ANY. I agree this should be filtered out. 

On your first point, If I understand correctly, you're suggest we change semantics such that:

- ALL fibers blocked waiting on a wildcard {MICROBIT_ID_NOTIFY_ONE, MICROBIT_EVT_ANY} event wake up when a matching NOTIFY event is raised.
- ONE fiber blocked on a *specific* {MICROBIT_ID_NOTIFY_ONE, X} event will wake when that NOTIFY event is raised.

If this is what you mean, i agree with you completely, as we're only tightening up the semantics for the edge case, which is currently ambiguous... Would you like to submit a PR to this effect?

For your second point, I *think* this is also a good suggestion, but I'd need to look at a complete patch to see any possible side effects... Again, fancy putting a PR together for this that we can look at?

drmpf, 2016-10-19:
Here is a patch for MicroBitFiber.cpp

[MicroBitFiber.cpp.zip](https://github.com/lancaster-university/microbit-dal/files/540381/MicroBitFiber.cpp.zip)  (This is actually a zip of the patch file NOT the cpp file)

I have modified my original suggestion slightly.  In the attached patch, MICROBIT_ID_NOTIFY_ONE is not a valid ID for wait_for .  It is only valid as a special event id to select just one waiting NOTIFY fibers to be released.