lancaster-university/microbit-dal #224: microbit DAL doesn't build on mbed 5

jaustin, 2016-10-11:
By default, the DAL doesn't build on mbed 5- there are a small number of API changes and build system changes that need to be accommodated

Some of these have been made by @tyomitch here:
https://github.com/tyomitch/microbit-dal

It is possible to use mbed 5 without the whole of mbed OS, and this would be a good step for micro:bit

jaustin, 2016-10-11:
@RidaJichi is also looking at aspects of this

RidaJichi, 2016-10-11:
Right @jaustin 
I managed to build a micro:bit Hello World for mbed os 5 by resorting to space optimization with using newlib-nano in mbed-os, beside the microbit-DAL changes introduced by @tyomitch in https://github.com/tyomitch/microbit-dal to make it compatible with mbed os 5. 
Before this 'nano' optimization, 'mbed compile' showed that libc occupied ~2k of memory space. After the optimization, memory space occupied drops to less than half k. 
Still, the stack size must be decreased below the default of 2k (1k is good), and BLE **must** be disabled as it is very memory hungry. BLE-specific code can also be hacked out from microbit (https://github.com/lancaster-university/microbit) as it can save ~1k for heap. 

In conclusion, here is the recipe for bringing micro:bit Hello World to life on mbed-os 5 for now:
1- Copy main.cpp file (from https://github.com/lancaster-university/microbit-samples/blob/master/source/main.cpp) and and add libraries to your mbed project
`mbed` add https://github.com/ARMmbed/mbed-os`
`mbed add https://github.com/tyomitch/microbit-dal`
`mbed add https://github.com/lancaster-university/microbit`
2- Add the linker option `"--specs=nano.specs"` in mbed-os/tools/profiles/default.json (under "GCC_ARM" ["ld"] assuming gcc_arm toolchain)
3- Change the stack size from 2k (default) to 1k, specify `ISR_STACK_SIZE`, and disable BLE when compiling:
`mbed compile -c -D __STACK_SIZE=1024 -D ISR_STACK_SIZE=1024 -D MICROBIT_BLE_ENABLED=0`
BLE can also be disabled by hard-resetting `MICROBIT_BLE_ENABLED` to 0 in microbit-dal/inc/core/MicroBitConfig.h

If the microbit application is single-threaded, the option `"-DMBED_RTOS_SINGLE_THREAD"` can be added in mbed-os/tools/profiles/default.json (under "GCC_ARM" ["common"] assuming gcc_arm toolchain).
If not, using newlib-nano can have implications on the microbit as it uses fibers rather than threads (still under investigation).

finneyj, 2016-10-19:
Thanks @jaustin @RidaJichi 

I'm very supportive of getting microbit-dal building against mainstream mbed, and resolving the issues we had to address in our local forks of the mbed repos along the way. 

If I understand correctly, there are two aspects to mbed 5. There's mbed os proper, which includes your RTOS scheduler etc. Then there's the SDK, which is pretty much similar to the mbed 2.0 HAL? Are we looking to support both these builds?

For the SDK builds, we should be looking to have an equivalent (or ideally better!) build than we currently have against mbed classic. 

For mbed-os proper, I presume you do want to support BLE though? This will be quite a significant aspect for your partners/customers won't it? So some thought required there methinks... Probably starting with an analysis of where the memory use is, and then perhaps a tweaked configuration of both mbed-os and microbit-dal to suit?

OwenBrotherwood, 2017-01-20:
Is there a summary of WIP (Work In Progress or Who In Progress)?

zerog2k, 2018-01-01:
Any progress on this? Also +1 on wanting mbedOS 5 support.
As I understand it, the hardware decision to use 16k nrf51822 severely limits available ram after the ble softdevice (s110? s130?) is using something like at least 9k+. But even without BLE support, keeping just the proprietary 2.4 GHz radio support (esb w/ gazell ?), seems still quite useful.

Is there any feature bounty on this which we could pitch in to?