lancaster-university/microbit-dal #225: Is it possible to set `BLEName` parameter to a custom value?

josephroberts, 2016-10-18:
Is it possible to set `BLEName` parameter to a custom value? Are there any plans to allow this - perhaps via a parameter in `config.json`?

finneyj, 2016-10-19:
Hi @josephroberts 

This is a good suggestion. The name is pre-defined for microbit.co.uk builds such that the micro:bit family of Android/IPhone apps (Samsung BLE app etc) can detect and program a micro:bit over BLE.

i think we should support other scenarios too though, and maintain the current name as a default. I'd be happy to have a CONFIG parameter for this. 

There was also some discussion when @Thomas101 added Eddystone functionality about creating a class to better handle BLE advertising / protocols using GAP as a transport, that this also relates to I think. Are there any/many compelling use cases for changing name at runtime?

kevinjwalters, 2020-06-29:
It looks like there's a ticket with lots of discussion covering setting the BLE name from the application in #306