lancaster-university/microbit-dal #226: radio.getRSSI() docs, no details of the actual returned result

whaleygeek, 2016-10-21:
The radio.getRSSI() method has no documentation as to the returned range, format and encoding, and interpretation of the RSSI value.

I note from this line in the source code:

https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitRadio.cpp#L238

it comes directly from the NRF radio structure.

I also note from this line in the source code:

https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitRadio.cpp#L90

that if the CRC in the packet fails, it sets the RSSI to 0 (i.e. the maximum possible signal level), implying there is no value that represents 'None' or 'no reading available' for RSSI.

Could we add some additional information in the comments of this method (or in the comments of the module as a whole) explaining what the valid range of return results of the RSSI value is and how to interpret them, including the case of 'no value available'? 

E.g. PXT for example uses 0 to mean strongest, and 255 to mean weakest. Is this the raw value returned from this DAL method, or is it translated?

I'd also like to know the mapping, if there is any, from RSSI to actual signal strength?

RSSI, by industry convention, is represented as 0 for the strongest signal (the reference signal), and as a negative number for values below that. What receive power level does 0 and 255 represent, (if anything)?

https://en.wikipedia.org/wiki/Received_signal_strength_indication

The value used by PXT is not completely in following with this convention (I think?), as in two's compliment, the number 255 represents -1 and 254 represents -2 (meaning that 255 is actually a bigger number than 254). whereas in PXT, 255 represents the weakest signal, and 254 represents a slightly stronger signal! So it seems it uses an inverted range, i.e. if you write numbers down on a sheet of paper with 0 at the top, then on each line below that you count 1, 2, 3... 255, then the number at the top of the paper is the strongest signal, and the number at the bottom of the paper is the weakest signal. Thus it's not a 2's complement representation of the RSSI in any shape or form.

I've also set up an issue request with the MicroPython guys to add RSSI to their radio API, and linked back to this issue to cross-link the discussions on RSSI, in the hope we can get some level of consistency in concept at least across all the editors that use the radio.

I remember some discussions about this months ago when PXT was surfacing RSSI, but I can't find those discussions at the moment to cross-link them into this issue.

finneyj, 2016-10-21:
@whaleygeek - thanks for noting the lack of documentation - will fix.

The value returned is -dbm, as you describe, but with an inversion. Where possible we always return SI values from microbit-dal (with the view that it will make it easy to support cross-curriculum learning across science subjects). The nordic chip provides values in -dbm so we forward that straight through.

*but* there's an inversion that I didn't pick up on in the first driver release. When the Nordic docs said it returns -dbm, it really meant it. i.e. if the RADIO module return 54, then that's -54 dbm... As this was already in use in PXT at this point, we couldn't easily change polarity without breaking something. I'd happily see this flipped, but we'd need downstream folks agree @pelikhan primarily.

Alternatively, we could introduce another method parameter / model to provide the inversion on request. Or, just correct this via documentation.

pelikhan, 2016-10-21:
We are deprecating the current set of radio apis for arguments in the data received event. So we could fix this in PXT.

finneyj, 2016-10-21:
perfect - let's take the opportunity to correct this then. I'll update the driver to return the accurate dbm value as an int value in the next release.

finneyj, 2016-10-25:
Additional documentation added on RSSI operations, and dbm value inverted as discussed.

closed 2016-10-25 by finneyj