lancaster-university/microbit-dal #233: Make BLE HID keyboard available

jaustin, 2016-11-01:
There's already a PR for this, but it is for an older version of the DAL and not ready for merging.
https://github.com/lancaster-university/microbit-dal/pull/117

@matthewelse did the original work

matthewelse, 2016-11-01:
@jaustin I've been meaning to look at this for quite a while 😏 . I've got quite a busy week or two ahead, but I'd definitely be interested in getting it to the point where it can be merged into the DAL. Do you have any specific time frame for getting this merged by?

jaustin, 2016-11-01:
@matthewelse that was quick! I'm actually just filing a bunch of issues to serve as possible 'up for grabs' tasks for interested people looking at the DAL... If you've not got the time to look at it now, no worries, but this issue will let us/anyone else interested discuss it, etc.

matthewelse, 2016-11-01:
@jaustin Makes sense - definitely if someone's interested in getting this ready I'm happy to work with them.

bluetooth-mdw, 2016-11-01:
I'd be happy to volunteer to document this as and when it's available. I haven't looked at the code in the PR but i assume / hope it's consistent with the approach taken for other services and as such will sit nicely alongside them.

matthewelse, 2016-11-01:
@bluetooth-mdw thanks - that sounds great.

bluetooth-mdw, 2016-11-01:
@jaustin fyi I have a few of the Bluetooth related issues on my list.....

jaustin, 2016-11-01:
@bluetooth-mdw Please just file as 'upforgrabs' if you've got other relevant issues :)

finneyj, 2016-11-01:
Awesome - thanks guys. have also been meaning to resurrect this for quite some time - it would be a great feature to have. Looking forward to seeing this one!

jamesadevine, 2016-12-06:
#249

loretod, 2018-09-20:
Greetings--- ran into this thread while looking for a Microbit HID solution. I'm particularly interested in using the bluetooth for assistive technology switch use. I mostly just need a user friendly and low cost way to send a space bar and tab. Any ideas for a noob?

masakjm, 2019-01-06:
> assistive technology

I am involved in assistive technology in volunteer activities in Japan.
We are publishing programs created with reference to the following pages.
https://os.mbed.com/teams/microbit/code/microbit_presenter/

https://os.mbed.com/users/masakjm/code/microbit_switch_if_3sw /

Please refer to the following site which I am writing. (I am sorry, only in Japanese.)
http://mahoro-ba.net/e2040.html
http://mahoro-ba.net/e2039.html