lancaster-university/microbit-dal #234: Scrolling possible on the edge connector?

jaustin, 2016-11-01:
I'd like to be able to scroll up/down by running my finger across the edge connector. It feels like this might be possible by doing some kind of capacitive sensing on P0,1,2 or more simply looking at shorting/touching of the other pins.

It might even be possible to interleave the testing of the 'shortedness' of the pins with the use of the display (like the light sensing is interleaved) and so using all the pins on the edge connector to detect scrolling.

finneyj, 2016-11-01:
Swipe to unlock / wake would be a really nice feature too. Would be really interesting to see if this can be done in the first instance - really asking for someone to do a "feasibility study" of a few different approaches. ;)

I think my first stab would be at driving the pins on the edge connector adjacent to P0, P1 and P2 ACTIVE LO then simply looking for a timed sequence of of the existing touch events to detect and LEFT-TO-RIGHT or RIGHT-TO_LEFT swipe.

pelikhan, 2016-11-02:
We already have issues with children shorting GND and 3v. Asking to swiper over the edge connector is asking for trouble. What happens when they use a piece of metal instead of their finger and hot short the edge connector?

jaustin, 2016-11-02:
@pelikhan I'm not sure I agree - we already invite kids to 'touch' P0,1,2, and this would be scrolling within that region. I guess there's some important education about "don't do this with metal", "don't do this the other end" and if we can't communicate that, then this is non-ideal. However, scroll is by its nature a transient thing, you don't scroll constantly, so we're not talking sustained, low resistance shorts. It's also a cool learning outcome if we can show shorting between pins as being the *way* we detect this.