lancaster-university/microbit-dal #236: Implement the 'radio' protocol over BLE

jaustin, 2016-11-01:
The radio protocol is awesome and perfect for classroom environments, but phones can't 'listen in'

It'd be great to have an option to run the radio over BLE using advertising packets and scanning. This would involve making S130 available for use by the DAL and then using the mbed scanning APIs. 

I imagine this task would involve working with @jaustin, @bluetooth-mdw and @pan-, as well as the DAL team.

bluetooth-mdw, 2016-11-01:
Happy to help with this one when the time is right

remay, 2016-11-01:
I can offer some help with getting the S130 running on the micro:bit.   It's memory usage is much higher than the S110.    I suspect that moving to S120 V2 softdevice is really where we want to end up with it's run-time configuration of memory depending on features used ...

jaustin, 2016-11-01:
@remay I agree - using the mbed 5 nRF5x port and S130v2 could give us much more configurability - see https://github.com/lancaster-university/microbit-dal/issues/224

finneyj, 2016-11-01:
Do we really need S130 just for scanning / advertising? Wonders if this GAP functionality is achievable on S110...

Agree completely about S130v2. In terms of SRAM consumption, it does look much better than S130v1 for central mode operation, but indications i've seen are that it's actually a bit *worse* than S110 for peripheral mode... which would give us a bit of a quandary...

bluetooth-mdw, 2016-11-02:
S110 won't allow you to scan so if the goal is to be able to support direct m:b to m:b wireless communication using Bluetooth, we're going to need S130. You'd need no GATT support at all though, only GAP and the radio protocol looks very simple. So there may be opportunities wrt special builds that reduce memory requirements here.

Btw, I suggest we rename "radio" as "radio over Bluetooth" doesn't sound right at all!. "micro:bit peer to peer" or "MBPP" anyone?.

That said... @jaustin you said that the disadvantage of the radio protocol is that it's not visible to phones. But I do wonder what use case would need this, given it's intended for micro:bit to micro:bit peer to peer comms. Do you have something in mind?

jaustin, 2016-11-02:
I guess I should s/phones/BLE devices in general/ - consider a RPi 3 and a number of micro:bits - at the moment we tend to use a micro:bit as a "Radio/MBP2P <--> serial" bridge. We'd be able to use that extra micro:bit for other exciting things if the Pi itself could scan, etc.

pelikhan, 2016-11-02:
getting BLE to work with rpi3 is dead easy unfortunately the event service is not implemented in node-BBC-microbit.

bluetooth-mdw, 2016-11-04:
@pelikhan I just submitted a PR to add support for subscribing to and receiving specific events from the event services:

https://github.com/sandeepmistry/node-bbc-microbit/pull/7