lancaster-university/microbit-dal #237: Default text scrolling should pause at the 'middle' of the character

jaustin, 2016-11-02:
The text scrolling on the micro:bit is constant rate at the moment.

However, it would be easier to read if instead, the text scrolled between characters, but each character paused when it was in the middle of the display.

I think this could be implemented as a new scroll function, but with an option in the MicroBitConfig to make it a default, perhaps?

jamesadevine, 2016-11-02:
@jaustin It would be better (in many ways), if the scroll/animation functionality could be defined as a pluggable interface to `MicroBitDisplay` rather than an additional function. That way different scroll functions (as you have described above) could be defined in an Animation object, which the display uses.

jaustin, 2016-11-02:
@jamesadevine that sounds nice - do you have time to outline in this bug what that would entail so someone coming browsing for interesting things to do can pick it up and run with it?