lancaster-university/microbit-dal #240: microbit-dal uses a custom version of the mbed library but that's not clearly explained

jaustin, 2016-11-07:
The DAL 'includes' mbed, either via the yotta dependency or using a .lib file when building in the online IDE.

However, this
a) Isn't really mbed convention, where 'mbed' is normally a top-level include
b) means that if you *start* from a normal mbed program and end up wanting to add the DAL, you get errors because there are two (incompatible) mbed versions.

The long term fix is to merge back the Lancaster changes, and we know that and are doing that.

In the short term, though, we should find some ways to clarify what's going on, and how to convert a 'normal' mbed program to use the DAL. 

Some possible ways to do this:
Restructure our online IDE examples to include 'mbed' at the top level, then have 'microbit', which depends on 'microbit-dal' which wouldn't have a .lib file include mbed-dev-bin.

If we did this as part of a larger update to our microbit examples, so that the newer structure is directly importable into the online IDE now that it supports git, it might actually be a net lower-hassle change. On the other hand, having the yotta dependency tree diverge from the one we use in the online IDE seeps suboptimal.

Thoughts @jamesadevine @finneyj.

(This is feedback from running a hackathon with mbed/microbit/DAL thanks to @LiyouZhou)

jamesadevine, 2016-11-07:
@jaustin All for making the process easier, and if moving the mbed import into microbit, rather than further down the tree is all that is required, seems like a no-brainer.

What do we have to do to get everything supported in the IDE?

OwenBrotherwood, 2017-01-04:
Could be "nice" for me at the present stage of micro:bit mbed then add DAL then work on python/javascript