lancaster-university/microbit-dal #242: Quadrature decoder driver

sh1boot, 2016-11-08:
Just wanted to mention that I've started fiddling with this, but I'll still listen to anyone who wants to tell me how to do it (or wants to tell me to stop doing it).

Will be doing this concurrently with a motor driver driver.  I don't know if that would also belong in here or in a separate module.

jamesadevine, 2016-11-08:
@sh1boot Sounds great!

You could take a look at some of our existing device drivers, specifically [the compass](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitCompass.cpp) and [the accelerometer](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitAccelerometer.cpp)
dal/blob/master/source/drivers/MicroBitCompass.cpp) and [the accelerometer](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitAccelerometer.cpp), and if you follow the model outlined there, it should be trivial to integrate your driver into the DAL. :smile:

All the drivers subclass [MicroBitComponent](https://github.com/lancaster-university/microbit-dal/blob/master/inc/core/MicroBitComponent.h), which can be used in conjunction with `system_timer_add_component` or `fiber_add_idle_component` to receive system (interrupt context) and idle callbacks (when the micro:bit has nothing to do).

sh1boot, 2016-11-21:
What sort of assumptions can I make about concurrent access to the class?  I can trivially block the direct hazard from `systemTick()`, but once it starts firing off events I'm no longer sure what those event handlers might expect to be able to do.

Do I just pick some operations I think are reasonable and make those safe and document them as such?

In the interim I just have a bunch of `__disable_irq()` and `__enable_irq()`.  There's not a lot of precedent for that approach in other modules, though.

drmpf, 2016-11-21:
I have a related question.  When micro:bit switches fibers, are all globals committed.  That is do you need to use volatiles to access variables between fibers, or not.

sh1boot, 2016-11-26:
@drmpf, I think the answer to your question is that the language itself implies that globals must be committed when you make any external function call (almost certainly including fibre-switching functions).  This is because the compiler can't predict whether or not that function it calls could end up depending on something you should have modified.

To speculatively answer my own question... Is it the case that `systemTick()` is run in its own fibre, and that by implication all other fibres are at a point in their lifecycle where they've _allowed_ themselves to be interrupted?

Of course that won't help me within an ISR; but I might otherwise assume that things like message handlers (if not launched from an ISR) cannot cut into the middle of a critical operation.  Is that right?

Or is another threading model also in play, here?

jamesadevine, 2016-11-26:
SystemTick will be called from interrupt context, there are no threading models in place, but higher level interrupts can interrupt your interrupt. 

Components are added to an array, and when the system timer interrupt is fired, the array is iterated over and systemTick is invoked.

@drmpf globals should be safe, and can be shared between fibers. 

Excuse my brevity, on mobile!

finneyj, 2016-11-29:
@sh1boot @drmpf @jamesadevine Sorry for slow response here, had a hectic few weeks.

You're all right. :-)

@sh1boot As we're running a non pre-emptive scheduler, it's safe to assume that your code will not be interrupted unexpectedly by another fiber. Threaded event handlers are also, by default, not re-entrant for this reason. Events will queue up if your haven't completed processing of the last one. 

You *could* theoretically get two fibers calling your class I guess, if you have a blocking call within your code. i.e. if one of your methods blocks the fiber for some reason (sleeps, waits for an event etc) then another fiber could call your code while the first is asleep.

You can't really protect against access from interrupt context though - no amount of locks will help you protect form that (as you can't block something in interrupt context)... which is why we go to such lengths to keep user code out of interrupt context wherever possible.

If you can share a draft of your code where you think you have potential race conditions, do share it and i'll take a look. Usually such issues can be avoided.

sh1boot, 2016-11-30:
I do have a [draft](https://github.com/sh1boot/microbit-dal/tree/qdec_driver) of my code, but after a couple of iterations against a [test](https://github.com/sh1boot/testqdec) application (PID control loop for motor with encoder) I realised that it's better to let the caller decide when and how to poll the hardware, rather than inserting myself into an interrupt context and trying to do it automatically.

I still have an option to use `systemTick()` to keep the accumulator up to date for simple use cases, but I believe I eliminated all the read-modify-write that could potentially conflict with interrupt context.

However, I do assume that an `int64_t` can be read or written atomically -- that is, that `strd` will be used, and that interrupt context runs to completion either entirely before or entirely after that instruction is executed.

The driver's "incomplete" in the sense that I'm not happy with the constructor parameters, and the comments might not be consistent, and the whitespace might not be regular, and I probably don't want to abbreviate the class name that way, and I need to test all of the configurations on the final version of the code...  All that long-tail stuff.

sh1boot, 2016-11-30:
Oh, and also I wasn't really sure whether I should take `MicroBitPin` references, or just `PinName`s.  Most drivers appear to take `PinName`, but they're referring to hardwired peripherals.  All the remaining pins are already owned by MicroBitIO.cpp, and I wasn't sure if a non-permanent connection to a breakout board would need to negotiate control with the MicroBit interface, or merely assume that the hardware will be in a fit-for-use state and go straight to the pins.

finneyj, 2016-11-30:
Thanks @sh1boot. Looking good. I like that you've take MicroBitPin references - will look clean when using your code. 

A PullRequest would be welcomed when you're ready, and we'll give you a friendly code review on the way in.

sh1boot, 2017-04-05:
Ping?

PR #247

closed 2017-06-13 by alennie1993