lancaster-university/microbit-dal #243: Why are Eddystone URL and UID mutually exclusive?

pelikhan, 2016-11-26:
It seems that using Eddystone URL or UID are exclusive? After glanching through the code, I don't see a reason why.

bluetooth-mdw, 2016-11-26:
You can't advertise both frame types in the same advertising packet. The Eddystone specification does allow interleaving different frame types in different advertising packets/events but not all hardware supports this capability and further more we're "keeping it simple" here.

pelikhan, 2016-11-26:
Why does it have to be a compile flag? Can't this be checked at runtime? Would be it possible to this sequence for example?

```
advertise url
stop adverstising
advertise uid
```

and still be ok?

pelikhan, 2016-11-26:
For reference: here is the PXT implementation (not functional since there is no DAL tag for this feature yet), https://github.com/Microsoft/pxt-microbit-eddystone

bluetooth-mdw, 2016-11-26:
The initial version from @Thomas101 introduced the compile flag.

I think it would be OK wrt the Eddystone spec to alternate between frame types by making run time calls but 99.99999% of the time people will not use it like that and compile flags are an attempt to keep things efficient I think.

pelikhan, 2016-11-26:
The compile flags are problematic for PXT as means having 2 separate repos for UID vs URI.

bluetooth-mdw, 2016-11-26:
@finneyj How would you feel about making the Eddystone support solely dependent on run-time calls rather than config? Fine by me, was just following the path already set and felt it might be worth trying to save some memory.... or can I rely on the compiler to do a good enough job if, say, I was not using the Eddystone URL type, only the UID type? Getting this stuff into PXT would be excellent.

@pelikhan That said we have the same situation with pairing choices (passkey vs just works vs none required) and that choice *cannot* be made from run-time calls afaik since pairing mode is not application code. So it might be as well to solve the PXT problem with variable config options if at all possible... this will come up again I suspect.

pelikhan, 2016-11-26:
Yes we have that limitation. The pairing is somewhat special - it should really be a "checkbox" in the script properties (at the project level) and not tied to a library. Separate discussion.

bluetooth-mdw, 2016-11-26:
Isn't it the same technical issue though? Both cases require the content of config.json to be varied according to something in your PXT application. Isn't that the limitation we face?

finneyj, 2016-11-29:
@bluetooth-mdw I'm happy to have runtime configuration of this sort of thing in general, as it reduces complexity and adds flexibility... You're quite right that we do need to keep an eye on code size of course - but I'd ideally like this to be at a fairly coarse grained level, where entire components (e.g. Eddystone) could be added or removed. Folks can then tailor their build accordingly if they run into memory limitations, but the basic builds are easy to use.