lancaster-university/microbit-dal #244: changing the magnetometer period does not work properly

bluetooth-mdw, 2016-11-27:
Using nRF Connect, if I write one of the documented, supported values to the magnetometer period characteristic and read it back, I get a different value. I believe values should be in little endian format, per the accelerometer period characteristic. Note that writing and then reading back from the accelerometer period characteristic does not suffer from this issue; the value read back is the value most recently written.

Examples:

1. Set the magnetometer period to 20 (0x1400 LE)

[image]

2. New value shown in nRF Connect UI

[image]

3. Read the value back... note value obtained

[image]

finneyj, 2016-11-29:
Hi @bluetooth-mdw.

I think this is correct behaviour. Looking at the Magnetometer driver, these are the sampling periods supported by the hardware:

```
const MAG3110SampleRateConfig MAG3110SampleRate[MAG3110_SAMPLE_RATES] = {
    {12500,      0x00},        // 80 Hz
    {25000,      0x20},        // 40 Hz
    {50000,      0x40},        // 20 Hz
    {100000,     0x60},        // 10 hz
    {200000,     0x80},        // 5 hz
    {400000,     0x88},        // 2.5 hz
    {800000,     0x90},        // 1.25 hz
    {1600000,    0xb0},        // 0.63 hz
    {3200000,    0xd0},        // 0.31 hz
    {6400000,    0xf0},        // 0.16 hz
    {12800000,   0xf8}         // 0.08 hz
};
```

So given 20ms, it chose the nearest it could manage - 25ms.

Maybe the docs need tweaking?

bluetooth-mdw, 2016-11-30:
Thanks. Yes, the documentation is wrong then. I'll fix.

bluetooth-mdw, 2016-11-30:
Just checking before I change the docs.... are the sample rates supported by the accelerometer the same as or different to the magnetometer? Right now the values documented for both are the same:

Accelerometer Period: Determines the frequency with which accelerometer data is reported in milliseconds. Valid values are 1, 2, 5, 10, 20, 80, 160 and 640.

Cheers

finneyj, 2016-11-30:
Looks different. The driver uses a broad selection of the available modes documented on page 20 of the datasheet: http://cache.freescale.com/files/sensors/doc/data_sheet/MAG3110.pdf