lancaster-university/microbit-dal #245: magnetometer bearing / heading affected by motion in the X/Y plane

bluetooth-mdw, 2016-11-27:
With the micro:bit pointing in approximately the same direction throughout the test, agitating it back and forwards along the X or Y axes, generating accelerometer data for those axes, also affects the magnetometer bearing value (delivered over Bluetooth) in a corresponding manner. Easier to see a graph of this so:

[image]

In fact you can see that 'shaking' along X, Y or Z affects the magnetometer bearing but the X direction seems to have the biggest impact.

Raw data from the test session attached too but I think this should be easy to recreate.

[2016-11-27-16-21-34-436.zip](https://github.com/lancaster-university/microbit-dal/files/614801/2016-11-27-16-21-34-436.zip)

I did try testing without Bluetooth but was not successful getting repeated readings from the compass to work.... presumably my code was incorrect:

```
int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();
    MicroBitI2C i2c(I2C_SDA0, I2C_SCL0); 
    MicroBitAccelerometer accelerometer(i2c); 
    MicroBitStorage storage; 
    MicroBitCompass compass(i2c, accelerometer, storage); 

    while (1) {
        int heading = compass.heading();
        uBit.serial.printf("heading :%d\n",heading);
        uBit.sleep(50);
    } 
 }
```

bluetooth-mdw, 2016-11-28:
Tested again, with the micro:bit on a flat surface so that motion along X and Y was more precisely controlled. Got essentially the same result, with X and Y motion affecting the heading/bearing signficiantly and Z affecting it only a little.

[image]

martinwork, 2016-11-28:
The draw-a-circle compass calibration routine uses the accelerometer values to enable the tilt of the microbit to be taken into account when calculating the compass heading from the xyz magnetometer values. I think motion will make that calculation behave as if microbit is tilted and so the tilt compensated compass heading is only valid when microbit is still.

bluetooth-mdw, 2016-11-28:
Thanks @martinwork 

Unfortunate if true. I had in mind capturing both accelerometer and directional data with the micro:bit mounted on a moving vehicle.... that use case is not possible it would seem.

martinwork, 2016-11-28:
If you can assume microbit won't be tilted, I think you could avoid the calibration by not passing accelerometer to the compass object constructor.

finneyj, 2016-11-29:
right - thanks @martinwork. I think you might be right here. Tilt compensation algorithm will definitely be thrown off if the device is under acceleration... 

@bluetooth-mdw would indeed be interesting to see if you can reproduce this with tile compensation disabled...

kevinjwalters, 2020-09-20:
Given that `math.degrees(math.atan2(compass.get_x(), compass.get_y()))` gives the correct answer on my micro:bit running `MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit v1.0.1 with nRF51822` it makes me wonder about `MicroBitCompass::basicBearing()`:

https://github.com/lancaster-university/microbit-dal/blob/96126fd23f3147257c62cd1aa98a26588c869d63/source/drivers/MicroBitCompass.cpp#L488-L501

That is using a conversion to `SIMPLE_CARTESIAN` which I think swaps `x` and `y`. The MP code is the same as the C++ code and gives correct answer but it doesn't perform that swap. What am I missing here?