lancaster-university/microbit-dal #248: advertiseEddystoneUrl connectable as true may not work

OwenBrotherwood, 2016-12-05:
The examples that I have seen for advertiseEddystoneUrl have had connectable as false.

Currently, no success with having it set to true and seeing a beacon

https://github.com/lancaster-university/microbit-dal/blob/12bf127ae28c93498a5d3cf50894866496d9d335/source/bluetooth/MicroBitBLEManager.cpp#L515

jamesadevine, 2016-12-05:
What does your config look like?

OwenBrotherwood, 2016-12-05:
Hi James,
I wish I could give a quick answer: the actual config is under the pxt-microbit layers so it is could be hard to find :( :: "I will try".

Before opening this issue, I wanted to produce something via https://developer.mbed.org/compiler/#nav:/; but have failed in finding a library that supports eddystone so far.
Any advice on a successful usage of mbed would be great to create a simple case to check the connectable parameter would be great, otherwise "I will try again"

pelikhan, 2016-12-05:
Working on dump module.json into the built/ folder in PXT to diagnose this kind of issues.

OwenBrotherwood, 2016-12-05:
And I am still failing with my efforts on mbed.
Found a team site, but so far nothing to import for a basic example of eddystone that just works
https://developer.mbed.org/teams/microbit/code/

jamesadevine, 2016-12-05:
Hey Owen,

I'm on mobile at the moment. Eddystone was only released on Tuesday, so it will take a couple of cycles to be picked up.

Have you tried yotta?

Best,

James.

https://developer.mbed.org/teams/microbit/code/

OwenBrotherwood, 2016-12-05:
I will try and get yotta to work:  currently on Windows 10 but I have a virtual machine I can power up.
I get back after a yotta experience using the on-line material for Windows and/or Linux
https://lancaster-university.github.io/microbit-docs/offline-toolchains/

OwenBrotherwood, 2016-12-05:
yotta Windows 10 "Hello world" went well.
However https://github.com/lancaster-university/microbit-samples/tree/master/source/examples/bluetooth-eddystone-url failed.
I copyed the example into the source folder and clean then build as per README.md'en

I will try some other examples while I am at it :)

```
C:/Users/owen/microbit-samples/source/main.cpp: In function 'void startAdvertising()':
C:/Users/owen/microbit-samples/source/main.cpp:12:21: error: 'class MicroBitBLEManager' has no member named 'advertiseEddystoneUrl'
     uBit.bleManager.advertiseEddystoneUrl(URL, CALIBRATED_POWERS[tx_power_level-1], false);
                     ^
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed

```

jamesadevine, 2016-12-05:
Hey Owen,

I'll look in a bit!

Best,

James.

jamesadevine, 2016-12-05:
Off the top of my head, have you copied config.json into the root of microbit-samples?

Best,

James.

OwenBrotherwood, 2016-12-05:
I want to repeat the procedure, but having the config.json in the root and not the source dir (together with main.cpp) gave no errors ... but I cannot find the hex either :) 

"I will try again" both with the eddystone url and one of the examples that has worked to make sure I have enough practise.

OwenBrotherwood, 2016-12-05:
https://github.com/lancaster-university/microbit-samples README.md does state the config.json should be in the root: I hadn't gotten that far in reading.
It probably triggers the inclusion of the correct libraries needed.

OwenBrotherwood, 2016-12-05:
true or false? 
both work
and the config used (from the example) is the same as the example.
Thanks James: I think it is back to the pxt-microbit board again ( I can see that pelikhan has merged some debugging help into the repro

----

config.json that works with true or false

```

{
    "microbit-dal": {
        "bluetooth": {
            "enabled": 1,
            "pairing_mode": 0,
            "private_addressing": 0,
            "open": 0,
            "whitelist": 0,
            "advertising_timeout": 0,
            "tx_power": 0,
            "dfu_service": 0,
            "event_service": 0,
            "device_info_service": 0,
            "eddystone_url": 1,
            "eddystone_uid": 0
        },
        "gatt_table_size": "0x600"
    }
}

```

closed 2016-12-05 by OwenBrotherwood

pelikhan, 2016-12-05:
Ø              "open": 0,

Does it work with open: 1?

OwenBrotherwood, 2016-12-05:
@pelikhan I will try with open:1

OwenBrotherwood, 2016-12-05:
@pelikhan I found a yotta.json in the bluetooth build of pxt-microbit which is probably now the answer to Jame's original question about what the config looks like, so I put it here for completeness:
```

{
  "config": {
    "microbit-dal": {
      "bluetooth": {
        "enabled": 1,
        "private_addressing": 0,
        "whitelist": 1,
        "advertising_timeout": 0,
        "tx_power": 6,
        "dfu_service": 1,
        "event_service": 1,
        "device_info_service": 1,
        "eddystone_url": 1,
        "eddystone_uid": 0,
        "pairing_mode": 1,
        "open": 0,
        "security_level": "SECURITY_MODE_ENCRYPTION_NO_MITM"
      }
    },
    "gatt_table_size": "0x700"
  },
  "dependencies": {
    "microbit": "lancaster-university/microbit#v2.0.0-rc7"
  }
}

```

OwenBrotherwood, 2016-12-05:
0 and 1 both work (also with true ar false)

jamesadevine, 2016-12-05:
@OwenBrotherwood @pelikhan For this to work, based on the PXT config provided above, whitelisting needs to be disabled. Otherwise, you will only be able to see the micro:bit's advertisements if you are bonded.

jamesadevine, 2016-12-05:
@OwenBrotherwood Could you summarise your findings? Does true/false work for Eddystone in the microbit-samples repo?

OwenBrotherwood, 2016-12-06:
Summary:

- True/False work in the sample
- One should always read the README.md where it is clearly stated the config.json should be in the root of the example structure to trigger inclussion of other libraries (my bad)

A big thanks for your help: it helped to quickly find where the problem lies.
Peli has enabled some debugging in the pxt-microbit project that allows quicker findings of problems.
I have now gotten a yotta installation local: which is good for ensuring where possible problems are to be found.
I will be testing interval and power from yotta to check it is good,

Note: I am a community contributor to pxt/pxt-microbit: my projects are on http://microbit-drill.org/ with The Great Treasure Hunt being a request from someone in Denmark for beacons in Blocky/pxt as the driving force for getting beacons to work.
https://github.com/microbit-drill/epxt encapsulates pxt-microbit to allow offline and pre-release use of features direct from MS github for pxt/pxt-microbit.

farooqhasny, 2017-07-10:
thanks @OwenBrotherwood placing config.json worked for me which yotta 
but now i am getting the same error for plateformio 
`src\main.cpp:16:21: error: 'class MicroBitBLEManager' has no member named 'advertiseEddystoneUrl'`
any ideas ??? 
much appreciated 

thanks

jamesadevine, 2017-07-10:
@farooqhasny We don't maintain the platformio fork, but I would imagine it is out of date. 😞