lancaster-university/microbit-dal #250: advertiseEddystoneUrl: length of url

OwenBrotherwood, 2016-12-06:
https://lancaster-university.github.io/microbit-docs/ble/eddystone/ gives an example of an url advertisement where https://goo.gl/TlUTF7 is presented to advertiseEddystoneUrl as the url to be advertised.

The code for MicroBitEddystone.cpp seems to indicate that the processing of redusing https:// down to one byte encoding of 0x03 leaving the remaining characters as characters,  goo.gl/TlUTF7 as there are no further url encoding possibilities left to reduce the number of bytes sent over the air.

Given that an URL can have an overhead of up to 8 char (https://) which is then reduced to the one byte encoding of 0x03, is the max length of the url for advertiseEddystoneUrl more than 17? or perhaps in another release 

https://github.com/lancaster-university/microbit-dal/blob/12bf127ae28c93498a5d3cf50894866496d9d335/source/bluetooth/MicroBitEddystone.cpp#L56

jamesadevine, 2016-12-06:
Hey Owen,

This is actually defined in the EddyStone spec:
https://github.com/google/eddystone/tree/master/eddystone-url

This is because the advertising packet supported by Bluetooth LE doesn't allow for such large amounts of data to be advertised. So there's not a lot we can do about this :(

Best,

James.

https://github.com/lancaster-university/microbit-dal/blob/12bf127ae28c93498a5d3cf50894866496d9d335/source/bluetooth/MicroBitEddystone.cpp#L56

OwenBrotherwood, 2016-12-07:
Hej James,
Yep: I had changed my initial enquiry after a slack talk and found the code that showed that the micro:bit handled url encoding: ie the calling program does not need to use byte codes for the url.

Without reading how the micro:bit implements the advertiseEddystoneUrl, one can be in doubt as to how much the dal calls.
It could be a good idea to add something to the doc of the call that the micro:bit implements the byte encoding such that the url length presented to to advertiseEddystoneUrl can be longer than 17 bytes given the encoding done by dal.

The dialog came about as the pxt-microbit documentation included that the url should be less that 18 bytes: possibly because of the 18 in the code which can be mistaken to be the max limit of the url allowed.
I made a pull request for the pxt-microbit documentation that needs to be improved: 18 changed to 17 but requires some statement about byte encoding of, for  example, https://www. that increases the size that is possible to be presented.

https://github.com/lancaster-university/microbit-dal/blob/12bf127ae28c93498a5d3cf50894866496d9d335/source/bluetooth/MicroBitEddystone.cpp#L56

>Maybe the page could include some short comments about the encoding of the url and that the encoding is handled by the micro:bit dal such that the size of the url presented to advertiseEddystoneUrl can be greater than 17

OwenBrotherwood, 2016-12-07:
I will see if I can find some correct and clear text that can be placed @ lancaster and in the reference section of pxt-microbit for the url.
https://github.com/Microsoft/pxt-microbit/blob/master/docs/reference/bluetooth/advertise-url.md#parameters

Thanks for all the help: at some stage it will mean the micro:bit can be used for easy beacon with block interface from https://pxt.io

OwenBrotherwood, 2016-12-07:
Could we work together to create a good text for https://lancaster-university.github.io/microbit-docs/ble/eddystone/ such that their is a heads up for what the url is?
I know it is described in the specification but it could be "nice" to give a heads up that the url is not just 17 bytes but with a "but" so that it is easy to understand and could be used as the base for pxt-microbit documentation.

jamesadevine, 2016-12-07:
@OwenBrotherwood I see, so your issue is that although 17 bytes is the maximum, it's not really the maximum as eddystone does optimisations underneath the hood, and this is confusing to users. A very good point.

Of course we would accept PRs to the microbit-docs repository: https://github.com/lancaster-university/microbit-docs

Although @bluetooth-mdw wrote the original documentation, so it would be great to have his input when he's not so busy.

OwenBrotherwood, 2016-12-07:
:) you got it in one.
I try and create something that can help in the meantime: it may not be great but could help for a while.

OwenBrotherwood, 2016-12-07:
https://github.com/lancaster-university/microbit-docs/pull/39

closed 2017-01-31 by jamesadevine