lancaster-university/microbit-dal #254: Pairing with Google Pixel fails

bluetooth-mdw, 2016-12-20:
Per the subject, pairing with the new Google Pixel always fails. See video demo of the problem here:

[https://www.youtube.com/watch?v=VrFWLcpe_-4&feature=youtu.be](https://www.youtube.com/watch?v=VrFWLcpe_-4&feature=youtu.be)

With assistance from Nordic support, this has been tracked down to the fact that the micro:bit disconnects abruptly, as soon as pairing has completed. This should not cause a problem but for some reason the Pixel seems to think the pairing process has not completed and so clears the keys relating to the micro:bit. The android bluetooth HCI snoop log shows that the controller sends a disconnect message to the host, telling it the remote device disconnected. Not long after, the host sends a command to the controller that it should delete the link keys for the microbit device. 

[image]

Commenting out the disconnect call here:

https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L493

"Solves" the problem i.e. pairing completes OK with no error.

Is this abrupt disconnect necessary? Obviously disconnection will occur when the micro:bit resets.

finneyj, 2016-12-20:
thanks @bluetooth-mdw. 

I seem to recall the disconnection is there to enable the storage of CCCDs. We should check, but i'm pretty sure that it's essential to store CCCD state for iOS devices before the micro:bit resets out of pairing mode. Storing the CCCD only proved possible during a disconnected state, hence the force disconnect... 

Based on code inspection, it looks like the disconnect is issued very quickly after pairing (to ensur eit happens before a user resets the micro:bit)... maybe simply waiting a little longer would suffice here? Is there an indication from the logs how long a delay is expected by the Pixel before a disconnect would be deemed ok?

bluetooth-mdw, 2016-12-20:
Nordic just recreated the issue on one of their dev boards and confirm a delay before disconnecting seems to resolve it. I'll try this (though I'm pretty sure I already did!) and let you know how it goes.

bluetooth-mdw, 2016-12-20:
PR submitted. Tested with Pixel. Does the job :-)

Kudos to Nordic for helping figure this out.

https://github.com/lancaster-university/microbit-dal/pull/255

bluetooth-mdw, 2016-12-22:
I just encountered what looks like the same problem with my Nexus 9 tablet with a hex file created a couple of months ago on a micro:bit. I'm beginning to think the issue may be with Android 7.1.1 which is what my Nexus is now running and what the Pixel is running too.

@pelikhan heads up as if I'm right we'll see this issue more often. Might be as well to get my fix into PXT soon.

pelikhan, 2016-12-22:
@finneyj is there a tag with the fix?

whaleygeek, 2017-01-04:
Anyone got a progress update on this one, as to when it might surface to live? Thanks!

DavidWhaleIET, 2017-01-10:
Has this fix been deployed yet does anyone know, @finneyj @jaustin

finneyj, 2017-01-10:
@DavidWhaleIET afraid this is blocked on me David - my bad. I'll get a release out this week for sure including all the BLE updates and PWM fix.

Unless anyone knows where I can get a machine to make a hoard of mini-me clones? maybe one at CES?

whaleygeek, 2017-02-02:
No problem. If we could tag this against a specific release future, I could monitor this myself without bugging you :)

DavidWhaleMEF, 2017-04-21:
Is this one going into rc9 do you know? Thanks!

pelikhan, 2017-04-21:
It should be in rc9. Try /beta.

DavidWhaleMEF, 2017-04-21:
Great thanks!

DavidWhaleMEF, 2017-04-21:
microbit.org-ticket: 472

DavidWhaleMEF, 2017-04-29:
Customer has reported that Google Pixel now pairs successfully with pxt.microbit.org/beta

@finneyj when the appropriate beta goes out to live, can you close this ticket down, ta.

Also shout out to @bluetooth-mdw for the many hours of head scratching and investigation that went into fixing this issue!

Well done team!

DavidWhaleMEF, 2017-04-29:
fixed by:
    pxt.microbit.org version: 0.12.5
    Microsoft MakeCode version: 0.12.63
    C++ runtime version: v2.0.0-rc9

Currently on pxt.microbit.org/beta

closed 2017-06-13 by alennie1993