lancaster-university/microbit-dal #257: Documentation for MicrobitDfuService?

kartben, 2017-01-09:
I would like to leverage the DFU feature of the micro:bit (which definitely works, when using e.g the Android app from Samsung), but I think I lack some information.
I manage to switch to the bootloader by writing 0x01 to the MicroBitDFUServiceControlCharacteristic, and then I have no luck uploading a package, and I think what I'm missing is how to properly generate an application update ZIP package using nrfutil. FWIW I'm using nRF connect android app for the actual upload.

Here's how I'm generating a package right now:

```
nrfutil pkg generate --application app.hex --application-version 1 --key-file private.pem --hw-version 51 --sd-req 0x67 app.zip
```

My understanding is that the app-version is not used but needs to be set, right? I'm not sure about the signing of the package, since I think the bootloader in the microbit is unsecured?

Any help would be most welcome :)
FWIW my plan is to write a tutorial on the topic, and maybe combine DFU with something like [Eclipse hawkBit](https://eclipse.org/hawkbit).

Thanks!

kartben, 2017-01-09:
Mmh interesting, uploading the app's .hex directly seems to work (yay!). I guess generating a *signed* package simply doesn't work because the bootloader explicitly expects unsigned code?

matthewelse, 2017-01-19:
@kartben since the version of the bootloader that we're using at the moment is a little bit older than the most recent version (without versioning/signing etc.), you should only need the raw hex file in order to successfully do a DFU. If you download the micro:bit app, there should be a demo hex file in there that I've used successfully in the past.

kartben, 2017-01-19:
Indeed. Thanks for the feedback @matthewelse.
I've done some good progress and I will be putting together a tutorial shortly. 

If you guys agree, I would leave this issue opened since I don't think the info available from https://lancaster-university.github.io/microbit-docs/ble/dfu-service/ (and in https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html as well) is good enough for a newcomer to use the DFU service without looking at the internals (arguably DFU is not for newcomers, but eh :-))

bluetooth-mdw, 2017-01-19:
https://lancaster-university.github.io/microbit-docs/ble/dfu-service/ needs to be updated to link to Nordic's documentation since DFU is defined by them.

jamesadevine, 2017-01-31:
@kartben it would be great if you could create a pull request based on your tutorial content to our docs repo: https://github.com/lancaster-university/microbit-docs