lancaster-university/microbit-dal #263: Documentation: clarification of the meaning of evt.timestamp

rharbird, 2017-01-28:
Hello,

This is a minor point relating to the documentation for io.eventOn.

The documentation [here](https://lancaster-university.github.io/microbit-docs/ubit/io/#eventon) states:

"MICROBIT_PIN_EVENT_ON_PULSE - Configures this pin to a digital input, and generates events where the timestamp is the duration that this pin was either HI or LO. (MICROBIT_PIN_EVT_PULSE_HI, MICROBIT_PIN_EVT_PULSE_LO)"

I believe that timestamp is, in fact, the interval since the last occurance of the event being triggered, it is not the duration that the PIN was in the state.

Rae

jamesadevine, 2017-01-31:
Hi @rharbird,

If I remember correctly, the timestamp after requesting `EVENT_ON_PULSE` is the duration of which the pin was HI or LO.

Here's an example of how that detection works:

| HI | LO | HI | LO | HI |
|--|---|----|----|---|
|  |PULSE_HI 200us| PULSE_LO 200us | PULSE_HI 200 us | PULSE_LO 200 us|

Where the pin state is represented across the top, and the events triggered in the row below.

The event is raised in interrupt context, and should therefore contain the timestamp accurate to the current time +10 us (an overhead from reading the Timer in mbed). 

I guess there is something which is suggesting that this behaviour is not what you're seeing, what is your scenario?

rharbird, 2017-01-31:
Hello,

Let me gather some data so that demonstrate what I think I'm seeing properly.

Rgds,

Rae

saxbophone, 2020-03-04:
Apologies for necro'ing an old thread, but as this issue is still open, I thought it'd be worth raising that the units used for this duration do not appear to be documented in the documentation available at https://lancaster-university.github.io/microbit-docs/ubit/io/#setdigitalvalue (I only discovered the units are µs and not ms by finding this thread).