lancaster-university/microbit-dal #267: Should UART characteristic table be member variable instead of stack var?

PaulAustin, 2017-02-09:
When connecting to a micro:bit that has the BLE UART service there were characteristics I did not recognize that showed up. I changed the table to be a member of the MicroBitUARTService class and the extraneous characteristics went away. The MBED library also uses a member variable.

```
  txCharacteristic = new GattCharacteristic(UARTServiceTXCharacteristicUUID, txBuffer, 1, txBufferSize, GattCharacteristic::BLE_GATT_CHAR_PROPERTIES_INDICATE);
 
    // This looks risky if charTable is used after the constructor is finished.
    GattCharacteristic *charTable[] = {txCharacteristic, &rxCharacteristic};

    GattService uartService(UARTServiceUUID, charTable, sizeof(charTable) / sizeof(GattCharacteristic *));
```

https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitUARTService.cpp#L85

It looks like the GattService constructor just holds onto the cahrTable that is passed to it. I might have missed something, though. 
Thank You.

martinwork, 2017-02-09:
On the face of it, the uartService object  doesn't live longer than charTable, so it should be OK. However, although addService doesn't store a pointer to uartService or copy its pointer to charTable, it looks like it copies all the pointers in charTable to a member variable array called p_characteristics.

So it seems that a pointer to rxCharacteristic will exist in BLEDevice after it goes out of scope. Most of the DAL GattCharacteristics  are created like rxCharacteristic. It's not clear to me exactly when this would a problem, but it certainly looks risky!

finneyj, 2017-02-09:
@martinwork @PaulAustin 

This is indeed one of the tricks we had to play to save some memory... it is walking on the dark side a bit, but for good reason. Every characteristic adds around 200 bytes of RAM overhead in the BLE stack (yes, really), and this really adds up when running any reasonable number of service. The characteristics are only ever used in an edge case by the mbed BLEDevice stack (for notifications/indications as I recall), hence we only keep the data "live" for those characteristics where this is enabled. note this is why txCharacteristic is heap allocated, but rxCharacteristic is stack allocated... and the same for the characteristic array.

I'm curious about @PaulAustin's issue and fix though. I also can't see why this would resolve the problem...

PaulAustin, 2017-02-09:
Thank you for looking at this. when I built the example unmodified from scratch. The BLE tool I used on my mac was listing some extraneous characteristics. The tool is "Adafruit Bluefruit LE Connect" Frankly they looked like random HEX strings. but I'm a bit new to this level of BLE so I ignored them.

Then I figured out the RX/TX swap. It's was a choice of fixing the client side (a JS based cordova app) I'm making , or the MBED-deivce side. I ultimately tried the MBED side. I have plenty of C++ experience so I know tracking down life-times with C++ can be hard at times, so I only half trusted my late night guess. But once I made the changes the extraneous characteristics disappeared, and the TX/RX switch worked. So I wanted to pass along the observation, especially since the MBED stack stored them as member variables as well. It really is impressive what you have been able to pack in there. Every byte counts.

I appreciate the details you noted, I'll go back and look more closely at my 'fixes' in light of your notes, later next week I expect. The preliminary changes can be seen in this fork:

https://github.com/PaulAustin/microbit-dal/commits/master

martinwork, 2017-02-12:
BLE_GATT_CHAR_PROPERTIES_INDICATE only occurs in the UART service, but _NOTIFY is used on quite a few stack allocated characteristics, so I ventured further into the dark side...

It turns out that ::write() is not called for the stack allocated NOTIFY characteristics outside the constructor - ::notify() is used, which does not refer to the ghost pointers.

I think a live GattCharacteristic is required:
(1) if setReadAuthorizationCallback or setWriteAuthorizationCallback is used
(2) if _INDICATE or _NOTIFY is specified and any of these is used to catch events for the characteristic: onUpdatesEnabled, onUpdatesDisabled or onConfirmationReceived.
(3) if _INDICATE or _NOTIFY is specified and ::write() is used.
(4) Strictly, when ::write() is called for any characteristic, but I think with plain READ/WRITE ones this will just mean an unnecessary NOTIFY or INDICATE call might be made.

Even if (2) is followed, I think there is a slim chance that any invalid pointer can cause a callback to receive spurious calls that have a valid handle value, which isn't the handle of the characteristic that generated the event. Currently only onConfirmationReceived is used - in the UART service.

Accelerometer, IOPin, Magnetometer and Temperature services call ::write() outside the constructor on a stack allocated READ/WRITE characteristic. Maybe localOnly==true should be passed for these?

It looks fairly simple to modify nRF5xGattServer to support (2) and (3) and (4) properly without using the p_characteristics pointer.