lancaster-university/microbit-dal #268: Bug in MicroBitImage.printImage

whaleygeek, 2017-02-11:
I was just trying to build a simple C++ app using your MicroBitImage::printImage example code, and discovered the following...

  * const uint8_t heart[] = { 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 1, 1, 1, 1, 1, 0, 1, 0, 1, 0, 1, 1, 1, 1, 1, 0, 1, 1, 1, 0, 0, 1, 1, 1, 0, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, }; // a cute heart
  * MicroBitImage i();
  * i.printImage(0,0,heart);

1. There is an error in this example:
The MicroBitImage.printImage() takes it's first two parameters as width and height (not offset onto the bitmap). So you are asking it to print an image of size 0,0 onto the existing bitmap.

2. The example code does not check the return type to be MICROBIT_INVALID_PARAMETER, so you would have never have seen this as a problem in your testing.

3. There is a typo in this line that validates the width and height parameters:

    // Sanity check.
    if (width <= 0 || width <= 0 || bitmap == NULL)
        return MICROBIT_INVALID_PARAMETER;

It should say (width <=0 || height <= 0 || ...

Even if you (correctly) pass in width=10, height=5 or whatever the dimensions of the image are...

4. You calculate the dimensions to copy here:

    // Calcualte sane start pointer.
    pixelsToCopyX = min(width,this->getWidth());
    pixelsToCopyY = min(height,this->getHeight());

So, because the default image on MicroBitImage::MicroBitImage (which calls init_empty) has a width=1 and height=1, and your width=10 height=5, only 1 pixel gets copied to the resultant canvas that was (from your example code) initialised using the default constructor to 'empty'

5. This then attempts to write to it's ptr, which was initialised to 'empty' by the default constructor. 
The declaration of 'empty' states the following:

static const uint16_t empty[] __attribute__ ((aligned (4))) = { 0xffff, 1, 1, 0, };

static const means you expect it to not change. But you have just changed it. On a platform that has a memory protection unit, this will cause a bus error, because you have now written to read only memory. On a platform that does not have a memory protection error but still loads this into RO memory (flash), the failure will go unnoticed. On a platform that actually loads this into RAM (uncommon these days) you have just changed the first and only pixel of your empty image.

6. The coding of your uint8_t initialisers assume that you are on a big-endian platform. If this code is ported to a little-endian platform, it will fail: 10, 0, 5, 0 is width=10 height=5 on a big-endian platform. On a little-endian platform it becomes width=2560, height=1280. It would be normal to guard endian-unsafe initialisers with a byte-swapping macro that chooses the correct byte order based on the endianness of the target platform.

/**
  * Constructor.
  * Create an image from a specially prepared constant array, with no copying. Will call ptr->incr().
  *
  * @param ptr The literal - first two bytes should be 0xff, then width, 0, height, 0, and the bitmap. Width and height are 16 bit. The literal has to be 4-byte aligned.
  *
  * @code
  * static const uint8_t heart[] __attribute__ ((aligned (4))) = { 0xff, 0xff, 10, 0, 5, 0, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 1, 1, 1, 1, 1, 0, 1, 0, 1, 0, 1, 1, 1, 1, 1, 0, 1, 1, 1, 0, 0, 1, 1, 1, 0, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, }; // a cute heart
  * MicroBitImage i((ImageData*)(void*)heart);
  * @endcode
  */
MicroBitImage::MicroBitImage(ImageData *p)

When you say 'literal' here, do you mean the whole 0xff,0xff,10,0,5,0,... or just the bitmap portion? If you mean just the bitmap portion, then that is _not_ 4-byte aligned due to the addition of your 0xFFFF reference count value. Presumably you want it 4-byte aligned for later efficient memcpy so I assume you mean the whole initialised structure, not just the bitmap part?

7. This whole example in the docs causes a compiler warning and two errors if I use your example code from the docs:

    MicroBitImage i();
    i.printImage(5, 5, heart);
    printf("%s", i.toString().toCharArray());

../src/main.cpp:21:20: warning: empty parentheses interpreted as a function
      declaration [-Wvexing-parse]
    MicroBitImage i();
                   ^~
../src/main.cpp:21:20: note: remove parentheses to declare a variable
    MicroBitImage i();
                   ^~
../src/main.cpp:22:5: error: base of member reference is a function; perhaps you
      meant to call it with no arguments?
    i.printImage(5, 5, heart);
    ^
     ()
../src/main.cpp:23:17: error: base of member reference is a function; perhaps
      you meant to call it with no arguments?
    trace("%s", i.toString().toCharArray());

finneyj, 2017-02-11:
Hi @whaleygeek 

Yep! Thanks for the detailed analysis!

Two problems here I think:

1) Validation is clearly wrong in the printImage() method. It should indeed test width and height as you describe - looks like a historic typo that no-one has tripped over before. 

2) Documented example is clearly wrong. Creating an emptyImage and then attempting top paste into it makes no sense. The example should be updated to create an image with a specified size... then the data structure will be in RAM, and will behave as you expect. 

So, one validation bug and one documentation bug I think... Agree?

whaleygeek, 2017-02-11:
Yep, that's the kiddies!

The analysis was detailed mainly because it surfaced from the bottom up, took me a while to trace what the real issue was and I kept notes so it made sense to give you the full story!

Yes, in my modified code I create a sized image of the right size, then printImage onto that and it works fine.

whaleygeek, 2017-02-11:
P.S. I was unclear as to whether the MicroBitImage i(); warnings and errors were compiler specific, and perhaps the compiler you build with is not as strict with respect to that 'ambiguity' (although the later errors are clearly errors?) - is this a docs issue again, should the empty constructor really be MicroBitImage i; instead of MicroBitImage i(); ?? In which case, I think that's a docs issue too.

jamesadevine, 2017-04-22:
@whaleygeek d5ee7d99b62becc4ceb6ed3725702d304939f5c9