lancaster-university/microbit-dal #270: Some files have no description of purpose at top

whaleygeek, 2017-02-12:
Most files that are c++ clearly have doc comments at the top explaining their purpose, which helps readers of the source code trying to learn about how it works to navigate and understand the purpose of that file. This is very useful.

However, the following files have no description of purpose:

bluetooth/ExternalEvents.h
bluetooth/MESEvents.h
bluetooth/MicroBitBLEManager.h
bluetooth/MicroBitEddystone.h

drivers/MicroBitFile.h
drivers/MicroBitFlash.h
drivers/TimedInterruptIn.h

asm/CortexContextSwitch.s
asm/CortexContextSwitch.s.armcc
asm/CortexContextSwitch.s.gcc

bluetooth/MicroBitBLEManager.cpp
bluetooth/MicroBitEddystone.cpp

drivers/MicroBitButton.cpp
drivers/MicroBitCompassCalibrator.cpp
drivers/MicroBitFile.cpp
drivers/MicroBitFileSystem.cpp
drivers/MicroBitFlash.cpp
drivers/MicroBitI2C.cpp
drivers/MicroBitSerial.cpp
drivers/MicroBitThermometer.cpp
drivers/TimedInterruptIn.cpp

types/PacketBuffer.cpp

finneyj, 2017-03-02:
Thanks again @whaleygeek.

Totally agree. I always try to do this to provide pointers to folks who are interested in looking a little deeper into "how stuff works". Would be great to be complete. Turning your list into a checklist here for tracking:

Some of these will have docs in a header file but not cpp file, so are an easy fi. Some probably need more detailed treatment. if anyone wants to help out here, contributions welcomed!

**Files Requiring Top level Comment Blocks:**

- [ ] bluetooth/ExternalEvents.h
- [ ] bluetooth/MESEvents.h
- [ ] bluetooth/MicroBitBLEManager.h
- [ ] bluetooth/MicroBitEddystone.h

- [ ] drivers/MicroBitFile.h
- [ ] drivers/MicroBitFlash.h
- [ ] drivers/TimedInterruptIn.h

- [ ] asm/CortexContextSwitch.s
- [ ] asm/CortexContextSwitch.s.armcc
- [ ] asm/CortexContextSwitch.s.gcc

- [ ] bluetooth/MicroBitBLEManager.cpp
- [ ] bluetooth/MicroBitEddystone.cpp

- [ ] drivers/MicroBitButton.cpp
- [ ] drivers/MicroBitCompassCalibrator.cpp
- [ ] drivers/MicroBitFile.cpp
- [ ] drivers/MicroBitFileSystem.cpp
- [ ] drivers/MicroBitFlash.cpp
- [ ] drivers/MicroBitI2C.cpp
- [ ] drivers/MicroBitSerial.cpp
- [ ] drivers/MicroBitThermometer.cpp
- [ ] drivers/TimedInterruptIn.cpp

- [ ] types/PacketBuffer.cpp

alennie1993, 2017-06-14:
Done a few, will commit soon. Left out FileSystem, might be better described by someone with more in-depth knowledge on how it actually works.