lancaster-university/microbit-dal #272: Documentation for MES Events

whaleygeek, 2017-02-12:
microbit.org-ticket: 620

Query from customer surfaced as a PXT issue, but this is clearly a docs issue in the DAL:

Essentially my question is - is and where is the documentation on 'events' in the PXT editor? The link provided only covers a few. There are other sources such as "MES_DEVICE_INFO_ID" that are not explained and also many of the "values" listed

The link I sent was this, but it doesn't contain any documentation about what the events mean, and misses out a huge number of events that are MES events in the PXT editor:

https://lancaster-university.github.io/microbit-docs/ubit/io/#message-bus-events

https://github.com/lancaster-university/microbit-dal/blob/master/inc/bluetooth/MESEvents.h

Where can we find documentation as to what each event means and how the user might be able to use it effectively for a useful purpose in the PXT editor? I am guessing that these are used to trigger actions at a smartphone, but do these actions only get triggered on Android, or are they also supported on iOS. Do they only get triggered if an app is running, or will they work regardless of what app is running on the smartphone?

Having experimented with some of these myself (and with support from Rae Harbird at UCL who was doing the same with PXT), I've found that some events can only be generated from some sources and not from others. Which sources can generate which events?

#define MES_REMOTE_CONTROL_ID               1001
#define MES_REMOTE_CONTROL_EVT_PLAY         1
#define MES_REMOTE_CONTROL_EVT_PAUSE        2
#define MES_REMOTE_CONTROL_EVT_STOP         3
#define MES_REMOTE_CONTROL_EVT_NEXTTRACK    4
#define MES_REMOTE_CONTROL_EVT_PREVTRACK    5
#define MES_REMOTE_CONTROL_EVT_FORWARD      6
#define MES_REMOTE_CONTROL_EVT_REWIND       7
#define MES_REMOTE_CONTROL_EVT_VOLUMEUP     8
#define MES_REMOTE_CONTROL_EVT_VOLUMEDOWN   9
#define MES_CAMERA_ID                       1002
#define MES_CAMERA_EVT_LAUNCH_PHOTO_MODE    1
#define MES_CAMERA_EVT_LAUNCH_VIDEO_MODE    2
#define MES_CAMERA_EVT_TAKE_PHOTO           3
#define MES_CAMERA_EVT_START_VIDEO_CAPTURE  4
#define MES_CAMERA_EVT_STOP_VIDEO_CAPTURE   5
#define MES_CAMERA_EVT_STOP_PHOTO_MODE      6
#define MES_CAMERA_EVT_STOP_VIDEO_MODE      7
#define MES_CAMERA_EVT_TOGGLE_FRONT_REAR    8
#define MES_ALERTS_ID                       1004
#define MES_ALERT_EVT_DISPLAY_TOAST         1
#define MES_ALERT_EVT_VIBRATE               2
#define MES_ALERT_EVT_PLAY_SOUND            3
#define MES_ALERT_EVT_PLAY_RINGTONE         4
#define MES_ALERT_EVT_FIND_MY_PHONE         5
#define MES_ALERT_EVT_ALARM1                6
#define MES_ALERT_EVT_ALARM2                7
#define MES_ALERT_EVT_ALARM3                8
#define MES_ALERT_EVT_ALARM4                9
#define MES_ALERT_EVT_ALARM5                10
#define MES_ALERT_EVT_ALARM6                11
#define MES_SIGNAL_STRENGTH_ID              1101
#define MES_SIGNAL_STRENGTH_EVT_NO_BAR      1
#define MES_SIGNAL_STRENGTH_EVT_ONE_BAR     2
#define MES_SIGNAL_STRENGTH_EVT_TWO_BAR     3
#define MES_SIGNAL_STRENGTH_EVT_THREE_BAR   4
#define MES_SIGNAL_STRENGTH_EVT_FOUR_BAR    5
#define MES_DEVICE_INFO_ID                  1103
#define MES_DEVICE_ORIENTATION_LANDSCAPE    1
#define MES_DEVICE_ORIENTATION_PORTRAIT     2
#define MES_DEVICE_GESTURE_NONE             3
#define MES_DEVICE_GESTURE_DEVICE_SHAKEN    4
#define MES_DEVICE_DISPLAY_OFF              5
#define MES_DEVICE_DISPLAY_ON               6
#define MES_DEVICE_INCOMING_CALL            7
#define MES_DEVICE_INCOMING_MESSAGE         8
#define MES_DPAD_CONTROLLER_ID              1104
#define MES_DPAD_BUTTON_A_DOWN              1
#define MES_DPAD_BUTTON_A_UP                2
#define MES_DPAD_BUTTON_B_DOWN              3
#define MES_DPAD_BUTTON_B_UP                4
#define MES_DPAD_BUTTON_C_DOWN              5
#define MES_DPAD_BUTTON_C_UP                6
#define MES_DPAD_BUTTON_D_DOWN              7
#define MES_DPAD_BUTTON_D_UP                8
#define MES_DPAD_BUTTON_1_DOWN              9
#define MES_DPAD_BUTTON_1_UP                10
#define MES_DPAD_BUTTON_2_DOWN              11
#define MES_DPAD_BUTTON_2_UP                12
#define MES_DPAD_BUTTON_3_DOWN              13
#define MES_DPAD_BUTTON_3_UP                14
#define MES_DPAD_BUTTON_4_DOWN              15
#define MES_DPAD_BUTTON_4_UP                16
#define MES_BROADCAST_GENERAL_ID            2000

DavidWhaleMEF, 2017-03-02:
I think the bigger issue might be, if they are listed in PXT (which I think they all are?) people will want to know what they do. So, either we document them and leave them in, or take them out?

Sorry if that sounds like I've got my support head on, but having things in the editors that do nothing just raise more questions. Thanks!

We could document them as 'for future expansion' of course, and document what they *might* be used for. That might be a good half way house for now.

finneyj, 2017-03-02:
Not at all - these are very good issues to raise. Looping in @pelikhan here for PXT view, as i think that's really a question closely related to both MEF and PXT.

I think it's an open question where the docs for something like this reside. microbit-dal is just a conduit here. It could act as a repo of well know / assigned numbers though - in which case a contributed page on our docs site would be the way to go.  

Thoughts?

DavidWhaleMEF, 2017-05-05:
Related PXT issue: https://github.com/Microsoft/pxt/issues/1340

DavidWhaleMEF, 2017-05-05:
Hi @finneyj sounds to me like we might need a page on our tech site, as this is clearly 'stuff in the middle' and 'how to use the plumbing provided'.

DavidWhaleMEF, 2017-05-05:
I've created a placeholder here for now. If you have additional information you would like me to copy/contribute into the article, let me know and I can do the heavy lifting here.

https://support.microbit.org/solution/articles/19000054469-mbtn3-mes-event-registry