lancaster-university/microbit-dal #273: BLE time client

OwenBrotherwood, 2017-02-17:
Hi
I wonder if there is room for a BLE time client such that the micro:bit/DAL can expose the real time relative to the world.
This would allow for applications that need some form of real time indication.

In short form, on power on, start asking for the time, with options for allways or just for a certain time period from any nearby time server that could be on a PC or mobil.

https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v8.x.x/doc/8.0.0/s110/html/a00064.html

OwenBrotherwood, 2017-02-17:
@bluetooth-mdw 
Hi Mr Bluetooth :) , 
Any thoughts about this?

finneyj, 2017-02-20:
Hi @OwenBrotherwood 

This is a great idea - An NTP style real time clock would definitely be useful. Once we have a snapshot, the rest would be fairly straightforward to implement. I'm not so sure this service is feasible though, as we cna only realistically run S110 at the moment, which gives us BLE peripheral capabilities, but not BLE client capabilities. I'm not 100% sure, but scanning the docs you referenced, it sounds like we'd need BLE client operation to enable this?

p.s. I think Mr Bluetooth is on a well deserved holiday at the moment...

OwenBrotherwood, 2017-02-20:
Just a short comment, on the nRF android (and other) sw from nordic, a time service is a couple of clicks to enablement

bluetooth-mdw, 2017-02-27:
The Nordic info does indeed describe the GATT client end of the standard Current Time Profile so we couldn't do this with S110.

finneyj, 2017-03-02:
Thanks @bluetooth-mdw. I'll close this one off for now then, as our current memory limitations won't let us realistically go beyond S110...

closed 2017-03-02 by finneyj

OwenBrotherwood, 2017-03-02:
micro:bit v52 :)