lancaster-university/microbit-dal #274: eddystone url reception

OwenBrotherwood, 2017-02-23:
I feel I have asked about this before, let's just make a seperate "issue" or wish.

The game (below) is to program a pxt to "see" another. My initial reaction was eddystone: and then I remembered something in a previous chat: not implemented due to constraints in dal.

Would be nice though.

[image]

pelikhan, 2017-02-23:
Use radio or file a feature request against the DAL.

OwenBrotherwood, 2017-02-23:
This is a request for DAL support of eddystone reception and not a pxt request 
Please reopen the issue

jamesadevine, 2017-02-23:
Hey @OwenBrotherwood can you elaborate a bit more on what you would like?

OwenBrotherwood, 2017-02-23:
Yes: I believe I had a short exhange on bluetooth with "Mr Bluetooth".
If I remember correctly, the ble profiles are not ment to be a full implementation (too little ram) but rather a subset in order to achieve ble with what is perceived as needed.

Now, with Eddystone being a future IoT, then sending from micro:bit is great.
Future fun stuff with reception of eddystone can give fun for kids with a bit of luck in play and learn.

Although radio and other methods are possible, and necessary with python, c++ can probably do it with the hardware out there if it is meant that the micro:bit becomes the key word with "I could do that with a micro:bit" instead of "I could do that with an Arduino".

finneyj, 2017-02-24:
Hi @OwenBrotherwood. I think I know what you mean here.

@pelikhan is right in that if you want direct micro:bit to micro:bit communication, the RADIO feature is great. very simple for kids to understand, and very easy for them to extend and tailor to their needs.

if I understand correctly, you're looking with the ability for a micro:bit to scan it's area for standard eddystone beacons (be they other micro:bits or not). This would be a useful feature I think, and *possible* even though we're only running as a peripheral. Eddystone uses GAP advertisements, which is a relatively simple protocol. This would be a nice additional feature.

OwenBrotherwood, 2017-02-24:
Yep: I allways agree in the end with Peli (he may not believe this)

https://microbit-micropython.readthedocs.io/en/latest/ble.html
- is also the same about ease of use of radio

However, things get "hairy" when one, start combining the education efforts of the pi and micro:bit foundations (charities or whatever).
Pi has ble on board now, and the clear explanation of areas of intersection for communication with BLE are still to be done.,

finneyj, 2017-03-02:
True - there's certainly a lot to do in this space. @bluetooth-mdw has done some great work on bluetooth gateways on the pi, and we now have support in node.js for the BLE services provided in this repo.

Would be reallycool to build on those foundations to develop cross platform learning resources. Huge potential there IMHO.

OwenBrotherwood, 2017-03-02:
@jaustin has something to think about now.