lancaster-university/microbit-dal #277: EVENT SERVICE characteristic MicroBit Requirements doesn't work

martinwork, 2017-03-05:
EVENT SERVICE characteristic MicroBit Requirements doesn't work.

When read, it always returns only a zero-length value. The problem is that EventModel::elementAt is not virtual, so the MicroBitMessageBus override is not called in onRequirementsRead.

The profile report says it returns "A variable length list" but doesn't mention that it returns one event at a time needing multiple reads to fetch the list.

Although documented in the profile report and created as supporting notify, looking at the code, I don't think it ever will send a notification.

I don't think polling it will work either because messageBusListenerOffset is only zeroed at construction and when BLE disconnects.

bluetooth-mdw, 2017-03-06:
Interesting. Don't think I ever tested reads.... primary use case is to use with notifications. Should indeed be a variable length list. Sounds like a bug.

martinwork, 2017-03-06:
Note this is about the Requirements characteristic and sending events to micro:bit not receiving notifications of client events from microbit.

The only call to notify is for microBitEventCharacteristicHandle, so I don't see how notifications can happen for microBitRequirementsCharacteristic.

The profile says: "The client should read this characteristic when it first connects to the micro:bit. It may also subscribe to notifications...". So I think read is the primary use case.

I think the main bug ( in EventModel::elementAt) is new to DAL v2.0.0.

I haven't checked, but I suspect the other bugs are not new:
- read only works once per connection
- notify never happens

finneyj, 2017-03-06:
@martinwork Thanks for the analysis - great attention to detail as always.

I think you might be right here. I introduced the EventModel abstraction to largely support the Python guys (as the MessageBus doesn't align with their needs, but it would be useful to still have a common concept of an event). 

I'll take a look to see if we can fix up that virtual reference without affecting any other code.

martinwork, 2017-03-15:
@finneyj elementAt only has this one caller.

All of the GattCharacteristics are created with len=0 and maxLen=4 so I don't think any will support "a variable length list" as specified in the profile. If that's true the code in onDataWritten is unnecessary and creating them with hasVariableLen=true might be unhelpful.

Maybe keeping it simple makes sense and it's the documentation that needs updating?

bluetooth-mdw, 2017-03-16:
No, this needs to support a variable length list, as specified, unless there's a show stopping reason which makes this impossible.

martinwork, 2017-03-16:
I don't think any of the EventService characteristics really behave as specified, probably for good reasons, though it would be possible. I took "variable length list" to mean transmitting more than one event in a single PDU. Have I misunderstood?

bluetooth-mdw, 2017-03-16:
Hi @martinwork , no you're spot on. Much of what the event service ended up being defined to be comes from way back in the day... 2 years ago now, when the DAL was relatively embryonic, and some of the "requirements" (i.e. crazy ideas) coming out of workshops, often concerning what the Samsung application might want to do, involved bidirectional exchanges of "requirements" and data and the idea had been to allow multiple event data structs to be exchanged in either direction in one Bluetooth notification or write PDU. Like many things in like, it seemed like a good idea at the time :-)

My guess, given the DAL message bus and how all that stuff works, is that collecting multiple events and packaging them in a single (e.g.) notification would in fact be..... tricky and perhaps that's why it's the way it is.

Perhaps a documentation update is in fact the way to go. Doubtless a requirement for multiple events in a single PDU will arise as soon as we do this, but that's the way it goes sometimes.

@finneyj what say you, oh wise one?