lancaster-university/microbit-dal #280: Turn on LEDs on display without flickering

pzoltowski, 2017-03-23:
Current smartphones have high frame rate Cameras (e.g. iphone 5s and above >120fps). I'm trying to do some realtime image processing for microbit pose estimation using OpenCV. I noticed though even if I turn on LEDs that are only on the same row, there is still noticable flickering when recording with >50fps Cameras. Is there some way to turn off scanning that is probably used for persistence of vision technique explained in documentation? Below the code I tested:
```
int main()
{
    MicroBitImage cross("0,255,0,255,0\n0,0,0,0,0\n255,0,255,0,255\n0,0,0,0,0\n0,255,0,255,0\n");
    uBit.display.print(cross);
    uBit.init();
    while(1) {
        uBit.sleep(5000);
    }
}
```

finneyj, 2017-03-23:
@pzoltowski 

Not so much I'm afraid! The LED matrix is wired in such a way that only 9 of the LEDs at any point in time can be illuminated. The whole display is strobed, and takes advantage of persistence of vision to make it look like 25 LEDs are on at the same time.

You're right that the display runs at round 50Hz (~55Hz i think). The frequency is a performance tradeoff - the higher the frequency, the more processor power it takes and more energy the CPU will suck from the batteries. hence we normally set the frequency to be just (but comfortably) outside that of human perception.

it is possible to change. if you don't mind building a program in C (and it looks like you don't!), you can just lower the scheduler frequency a bit - the display frequency is locked to that in this current version.

https://github.com/lancaster-university/microbit-dal/blob/master/inc/core/MicroBitConfig.h#L143

if you make the scheduler tick at, say, 2ms rather than 6ms, you'l be running at more like 165Hz, which should be good for your iPhone... 

You can also set it in your config.json if you're building using the yotta tool:
https://lancaster-university.github.io/microbit-docs/advanced/#compile-time-options-with-yotta

-- Joe

pzoltowski, 2017-04-01:
Thanks @finneyj for the tip. I changed  system tick period to even 1ms however It seems it helped only little bit - now can only record at maximum ~70Hz without observing flickering. I guess I will have to stick with printed markers as alternatives if I want to track fast movement, which is too bad because I see many useful application of microbit together with smartphones. Unless there is some other low level workaround (I don't mind diving deep into the microbit-dal code). I used the following  setting in config.json:
  
`"system_tick_period": 1`

jamesadevine, 2017-04-01:
@pzoltowski I've looked into this, the option doesn't appear to apply from `config.json`, requires further investigation.

Alternately, your program could look like this:

```cpp
#include "MicroBit.h"

MicroBit uBit;

int main()
{
    MicroBitImage cross("0,255,0,255,0\n0,0,0,0,0\n255,0,255,0,255\n0,0,0,0,0\n0,255,0,255,0\n");
    uBit.display.print(cross);
    system_timer_set_period(2);
    uBit.init();
    while(1) {
        uBit.sleep(5000);
    }
}
```

No jitter on my 6 plus with that program.

pzoltowski, 2017-04-01:
awesome @finneyj . Works perfectly!

jamesadevine, 2017-04-22:
The option was taking effect actually (just tested this again), perhaps @pzoltowski did the same thing as i did:

```json
{
    "system_tick_period": "1"
}
```

which should actually be:

```json
{
    "microbit-dal": {
        "system_tick_period": "1"
    }
}
```

closed 2017-04-22 by jamesadevine

pzoltowski, 2017-04-24:
correct @jamesadevine . I tried the same settings:
```
{
    "system_tick_period": "1"
}
```