lancaster-university/microbit-dal #281: set brightness causes micro:bit crash

DavidWhaleMEF, 2017-03-24:
re: https://github.com/Microsoft/pxt/issues/1743

The attached script crashes the micro:bit

I run this code on my micro:bit and it crashes the micro:bit

It displays a heart then it displays this pattern

.....
.....
.x...
.xxx.
.....
[microbit-brightness.hex.zip](https://github.com/lancaster-university/microbit-dal/files/868225/microbit-brightness.hex.zip)

pxt.microbit.org version: 0.9.55
Microsoft MakeCode version: 0.12.1
C++ runtime version: v2.0.0-rc8

finneyj, 2017-03-24:
Thanks @DavidWhaleMEF 

Thanks for simple repro case. Easily reproducible with this C code. 

```

finneyj, 2017-03-24:
Actually, after much head scratching, this is a not a crash at all. :-)

The driver did however prematurely optimize the condition where brightness=0, and left some LEDs illuminated in this case... it just looks like a crash.

Fixed in master with this commit:

https://github.com/lancaster-university/microbit-dal/commit/2f32db3e14a6b8f8f92ac9bdbcbb3dc98a914fc1

closed 2017-03-24 by finneyj

DavidWhaleMEF, 2017-04-18:
This fix isn't yet deployed into the live editors yet, is it?

Microbit.org-ticket: 1253 using blocks we still see this issue:

hexes attached for repro case.

[brightness.zip](https://github.com/lancaster-university/microbit-dal/files/928137/brightness.zip)

DavidWhaleMEF, 2017-04-18:
Just logged a k-base about this:

https://support.microbit.org/solution/articles/19000053041-set-brightness-0-does-not-work

P.S. Just a note 'fixed in master' doesn't mean it's 'fixed for user'. We need to get these fixes deployed to the live editors in order to completely fix the issue. Thanks!

finneyj, 2017-04-18:
@DavidWhaleMEF 

Correct - this is fixed in master, but not yet in a tagged release so won't be in PXT just yet.

Let's aim to scoop up a few more of the outstanding issues into a release in the next few days.

DavidWhaleMEF, 2017-05-02:
ping - just checking, did this make it into the last scoop or not do you know? Ta!

finneyj, 2017-05-03:
@DavidWhaleMEF 

Yes, I believe so. It was in microbit-dal-v2.0.0-rc9.

DavidWhaleMEF, 2017-05-03:
Can you link this ticket to the PR that put that into RC9 please, for traceability reasons. Thanks.

finneyj, 2017-05-03:
Sure - it was a very ,minor change so no PR, but was fixed in the commit cited above (https://github.com/lancaster-university/microbit-dal/commit/2f32db3e14a6b8f8f92ac9bdbcbb3dc98a914fc1)

DavidWhaleMEF, 2017-05-03:
Okay, so this is not fixed for the user yet as rc9 hasn't made it into live yet, but I see that rc9 is in PXT beta. I will get the customer to re-test their script and report back if there are any issues. Thanks for your help here!

finneyj, 2017-05-03:
no problem.