lancaster-university/microbit-dal #282: Invalid version specification errors when building using Yotta

nvgeele, 2017-04-05:
I am trying to build the examples from the `lancaster-university/microbit-samples` repository using Yotta (as described [here](http://www.forward.com.au/pfod/microbit/compileSetup.html)), but I can not build the project. I have pasted the output of the `yt build` command etc below.

I have no experience with Yotta, but it seems that the issue here lies with the `microbit-dal` dependency.

```
paperweight :: ~/tmp/microbit-samples ‹master› » git pull
Already up-to-date.
paperweight :: ~/tmp/microbit-samples ‹master› » git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working tree clean
paperweight :: ~/tmp/microbit-samples ‹master› » yt target bbc-microbit-classic-gcc
info: get versions for bbc-microbit-classic-gcc
info: download bbc-microbit-classic-gcc@0.2.3 from the public module registry
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.3 from the public module registry
paperweight :: ~/tmp/microbit-samples ‹master*› » yt build
info: get versions for microbit
info: download microbit@v2.0.0-rc8 from GitHub lancaster-university/microbit
info: get versions for microbit-dal
info: download microbit-dal@v2.0.0-rc8 from GitHub lancaster-university/microbit-dal
error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
error: Failed to satisfy dependencies of /Users/nvgeele/tmp/microbit-samples/yotta_modules/microbit-dal:
error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
error: mbed-classic not available
error: ble not available
error: ble-nrf51822 not available
error: nrf51-sdk not available
error: Missing dependencies prevent build. Use `yotta ls` to list them.
paperweight :: ~/tmp/microbit-samples ‹master*› » yotta ls                                                        1 ↵
error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
microbit-samples 2.0.0-rc8
┗━ microbit 2.0.0-rc8
  ┗━ microbit-dal 2.0.0-rc8 yotta_modules/microbit-dal
    ┣━ mbed-classic lancaster-university/mbed-classic#microbit_hfclk+mb6 missing
    ┣━ ble lancaster-university/BLE_API#v2.5.0+mb3 missing
    ┣━ ble-nrf51822 lancaster-university/nrf51822#v2.5.0+mb7 missing
    ┗━ nrf51-sdk lancaster-university/nrf51-sdk#v2.2.0+mb4 missing```

finneyj, 2017-04-05:
@nvgeele 

Curious - as a guess it's implying that some of our tags are not legal semver names.

Which version of yotta do you have? (```yotta --version```)

jamesadevine, 2017-04-05:
And also what operating system?

finneyj, 2017-04-05:
@nvgeele  

p.s. as a workaround you should be able to clone those three repos manually, as illustrated below. Please note the need to rename the BLE_API repo at the end if you take this route.

--joe

```
cd yotta_modules
git clone https://www.github.com/lancaster-university/mbed-classic
git clone https://www.github.com/lancaster-university/BLE_API
git clone https://www.github.com/lancaster-university/nrf51822
git clone https://www.github.com/lancaster-university/nrf51-sdk

mv BLE_API ble
```

DasPez, 2017-04-06:
I have exactly the same problem - Windows 10 x64 - Yotta 0.18.0 - installed using "yotta_install_v0.2.3.exe"

finneyj, 2017-04-06:
hmm - thanks @DasPez. My guess is these later version of yotta are over-validating somewhat, so not downloading the repos correctly. Does the above workaround work for you?

DasPez, 2017-04-06:
(workspace) C:\microbit\microbit-samples>yt build
    error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
    error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
    error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
    error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
    error: Failed to satisfy dependencies of C:\microbit\microbit-samples\yotta_modules\microbit-dal:
    error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
    error: Invalid version specification: "lancaster-university/BLE_API#v2.5.0+mb3"
    error: Invalid version specification: "lancaster-university/nrf51822#v2.5.0+mb7"
    error: Invalid version specification: "lancaster-university/nrf51-sdk#v2.2.0+mb4"
    error: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"
    Fatal Exception, yotta=0.18.0
    Traceback (most recent call last):
      File "C:\yotta\python\Lib\runpy.py", line 162, in _run_module_as_main
        "__main__", fname, loader, pkg_name)
      File "C:\yotta\python\Lib\runpy.py", line 72, in _run_code
        exec code in run_globals
      File "C:\yotta\workspace\Scripts\yt.exe\__main__.py", line 9, in <module>
      File "c:\yotta\workspace\lib\site-packages\yotta\main.py", line 61, in wrapped
        return fn(*args, **kwargs)
      File "c:\yotta\workspace\lib\site-packages\yotta\main.py", line 46, in wrapped
        return fn(*args, **kwargs)
      File "c:\yotta\workspace\lib\site-packages\yotta\main.py", line 244, in main
        status = args.command(args, following_args)
      File "c:\yotta\workspace\lib\site-packages\yotta\build.py", line 46, in execCommand
        status = installAndBuild(args, following_args)
      File "c:\yotta\workspace\lib\site-packages\yotta\build.py", line 109, in installAndBuild
        test = True
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 526, in getDependenciesRecursive
        test = test
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 451, in __getDependenciesRecursiveWithProvider
        _processed = _processed
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 451, in __getDependenciesRecursiveWithProvider
        _processed = _processed
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 430, in __getDependenciesRecursiveWithProvider
        test = test
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 330, in __getDependenciesWithProvider
        satisfyDep, specs
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 308, in satisfyDep
        if r and not sourceparse.parseSourceURL(dspec.versionReq()).semanticSpecMatches(r.getVersion()):
      File "c:\yotta\workspace\lib\site-packages\yotta\lib\sourceparse.py", line 114, in parseSourceURL
        raise InvalidVersionSpec("Invalid version specification: \"%s\"" % (source_url))
    yotta.lib.sourceparse.InvalidVersionSpec: Invalid version specification: "lancaster-university/mbed-classic#microbit_hfclk+mb6"

No

jamesadevine, 2017-04-06:
@DasPez @finneyj @nvgeele I've filed an issue with the yotta folks(https://github.com/ARMmbed/yotta/issues/803), it may be worth downgrading for now.

finneyj, 2017-04-06:
thanks @jamesadevine.

@DasPez, as a further workaround,in addition to manually pulling the other repos from github, you could also try editing  ```yotta_modules/microbit-dal/module.json``` and removing the tags from the dependencies listed. Such that your module.json looks a bit like this:

```
{
  "name": "microbit-dal",
  "version": "2.0.0-rc8",
  "license": "MIT",
  "description": "The runtime library for the BBC micro:bit, developed by Lancaster University",
  "keywords": [
    "mbed-classic",
    "microbit",
    "runtime",
    "library",
    "lancaster",
    "University"
  ],
  "author": "Joe Finney <j.finney@lancaster.ac.uk (mailto:j.finney@lancaster.ac.uk) >",
  "homepage": "https://github.com/lancaster-university/microbit-dal/",
  "dependencies": {
    "mbed-classic": "lancaster-university/mbed-classic",
    "ble": "lancaster-university/BLE_API#v2.5.0",
    "ble-nrf51822": "lancaster-university/nrf51822",
    "nrf51-sdk": "lancaster-university/nrf51-sdk"
  },
  "extraIncludes": [
    "inc/core",
    "inc/types",
    "inc/drivers",
    "inc/bluetooth",
    "inc/platform"
  ]
}
```

DasPez, 2017-04-06:
`"ble": "lancaster-university/BLE_API#v2.5.0",` (from your proposed module.json)
i removed '#v2.5.0' as well - now the build starts but ninja fails

    (workspace) C:\microbit\microbit-samples>yt build
    info: get versions for ble-nrf51822
    info: download ble-nrf51822@v2.5.0+mb6 from GitHub lancaster-university/nRF51822
    info: generate for target: bbc-microbit-classic-gcc 0.2.3 at C:\microbit\microbit-samples\yotta_targets\bbc-microbit-classic-gcc
    GCC version is: 4.9.3
    -- The ASM compiler identification is GNU
    -- Found assembler: C:/yotta/gcc/bin/arm-none-eabi-gcc.exe
    suppressing warnings from ble-nrf51822
    suppressing warnings from nrf51-sdk
    suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
    -- Configuring done
    -- Generating done
    -- Build files have been written to: C:/microbit/microbit-samples/build/bbc-microbit-classic-gcc
    [163/163] Linking CXX executable source/microbit-samples
    FAILED: cmd.exe /C "cd . && C:\yotta\gcc\bin\arm-none-eabi-g++.exe   -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -T"C:/microbit/microbit-samples/yotta_targets/bbc-microbit-classic-gcc/CMake/../ld/NRF51822.ld" -Wl,-Map,source/microbit-samples.map -Wl,--start-group source/CMakeFiles/microbit-samples.dir/C_/microbit/microbit-samples/source/main.cpp.o  ym/microbit/source/microbit.a ym/microbit-dal/source/microbit-dal.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/ble/source/ble.a ym/ble-nrf51822/source/ble-nrf51822.a ym/nrf51-sdk/source/nrf51-sdk.a ym/mbed-classic/existing/mbed-classic.a -lnosys  -lstdc++ -lsupc++ -lm -lc -lgcc -lstdc++ -lsupc++ -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o source/microbit-samples && cmd.exe /C "cd /D C:/microbit/microbit-samples/build/bbc-microbit-classic-gcc/source && arm-none-eabi-objcopy -O ihex microbit-samples microbit-samples.hex && srec_cat C:/microbit/microbit-samples/yotta_targets/bbc-microbit-classic-gcc/CMake/../bootloader/BLE_BOOTLOADER_RESERVED.hex -intel C:/microbit/microbit-samples/yotta_targets/bbc-microbit-classic-gcc/CMake/../softdevice/s110_nrf51822_8.0.0_softdevice.hex -intel microbit-samples.hex -intel -o microbit-samples-combined.hex -intel --line-length=44 && cd /D C:/microbit/microbit-samples/build/bbc-microbit-classic-gcc/source && C:/yotta/gcc/bin/arm-none-eabi-objcopy.exe -O binary microbit-samples microbit-samples.bin""
    ym/microbit-dal/source/microbit-dal.a(MicroBitFlash.cpp.o): In function `MicroBitFlash::MicroBitFlash()':
    C:/microbit/microbit-samples/yotta_modules/microbit-dal/source/drivers/MicroBitFlash.cpp:50: undefined reference to `btle_set_user_evt_handler'
    collect2.exe: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    error: command ['ninja'] failed

thats another topic and should be easy to fix, right?

finneyj, 2017-04-06:
indeed - these are all simply down to the versioning being incorrect... maybe your ble version is a bit out of date due to the typo you fixed above....

drop down into your ```yotta_modules/ble``` directory, and ```git checkout v2.5.0+mb3```

Any better?

DasPez, 2017-04-06:
No - still the same error

finneyj, 2017-04-06:
it's definitely in there. :-)

Looking at my repo here, this symbol is defined in the "nrf51822" repo.  Can you check that's also at the head revision?

nvgeele, 2017-04-06:
@finneyj @jamesadevine Yotta version 0.18.0, and I'm on macOS 10.12.3

Tried pulling the modules manually, changed `microbit-dal/module.json` file, and did the git checkout in `yotta_modules/ble`. I can also report having the same error as @DasPez is encountering.

finneyj, 2017-04-06:
@nvgeele @DasPez  

did you get checkout the HEAD revision of nrf51822 also? that's the repo where  this funciton is implemented..

DasPez, 2017-04-06:
yes

    (workspace) C:\microbit\microbit-samples\yotta_modules\nrf51822>git log -1
    commit 040baa12c08b439f9a09f0c1594ac625e58e2ad1
    Author: Joe Finney <joe@comp.lancs.ac.uk>
    Date:   Wed Mar 1 16:53:28 2017 +0000

        Add NRF event loop handler hook.

thegecko, 2017-04-06:
We've just published yotta 0.18.1 with a fix for this issue. Please let us know if it has worked.

nvgeele, 2017-04-06:
@finneyj Also yes.

@thegecko I upgraded to 0.18.1, thoroughly cleaned my project to make sure there were no traces of any intermediate files of the previous version, and now it works and successfully compiles.

DasPez, 2017-04-06:
Everything works perfectly!

finneyj, 2017-04-06:
great stuff - thanks @thegecko. :-)

nvgeele, 2017-04-07:
I guess this issue can be closed now?

jamesadevine, 2017-04-07:
Yes @nvgeele, closing.

closed 2017-04-07 by jamesadevine