lancaster-university/microbit-dal #284: ManagedString construct for char incorrect.

tballmsft, 2017-04-18:
ManagedString((char)code) creates a string of length 0 when code=0.

tballmsft, 2017-04-18:
I expect we have the same bug in CODAL (same code).

jamesadevine, 2017-04-22:
@tballmsft This issue is subtly complex, should this behaviour actually be allowed? Across `ManagedString` this behaviour is actively discouraged, even with our `char*` `length` constructor:

```cpp
if (str == NULL || *str == 0 || length == 0 || (uint16_t)length > strlen(str)) // XXX length should be unsigned on the interface
{
        initEmpty();
        return;
}
```
What this would mean is that we would sacrifice validation across the board by allowing strings containing 0's (given that 0 also doubles as a NULL terminator, making validation nigh on impossible).

 I think what we're talking about here is the difference between a byte buffer and a string, although 0 is a completely valid char, it's not a valid string. So isn't this what you would expect as a user?

tcurdt, 2018-02-09:
I just came across a similar issue in another constructor.
```
ManagedString::ManagedString(const char *str, const int16_t length)
{
    // Sanity check. Return EmptyString for anything distasteful
    if (str == NULL || *str == 0 || length == 0 || (uint16_t)length > strlen(str)) // XXX length should be unsigned on the interface
    {
        initEmpty();
        return;
    }
```
Here the `strlen` also adds the requirement for the `str` to be null terminated. Not exactly what I'd expected when I pass the length as argument.

One could argue that if `str` is shorter than the specified `length` it would be OK to stop at the null byte. Returning an empty string in that case does not make much sense to me.