lancaster-university/microbit-dal #285: Extra bytes prepended to MicroPython radio payloads?

DavidWhaleMEF, 2017-04-21:
A call for @finneyj here to help resolve this discrepancy.

Joe, can you let us know what these bytes are for, and whether they should or should not be there?

Sending strings from PXT to MicroPython is a bit of a nightmare at the moment, and it seems that a breaking change may have been done a while ago that has made it worse, can we dig into this and work out what is going on? Thanks!

https://github.com/bbcmicrobit/micropython/issues/438

DavidWhaleMEF, 2017-04-21:
@finneyj A repro test case is in this ticket (read the whole ticket, as our understanding of what the issue is morphs a bit towards the end!)

https://github.com/Microsoft/pxt/issues/1935

finneyj, 2017-04-21:
hi @DavidWhaleMEF 

Just for completeness, pasting in my reply from the micropython repo:

No breaking changes to this on microbit-dal side. I think PXT folks may have added an additional optional wrapper however that includes some metadata, which could be the source of the issue.

I knew that once a raw radio interface like this started getting used then folks would want to experiment with their own protocols, which is a good thing IMHO, so added three byte to (ironically) help manage interoperability. The three bytes are a protocol id (1 for basic raw datagram), version number (as defined by the protocol itself) and a group number. This is the 01 00 01 you're seeing there - it's a raw datagram version 1 frame in group 0.

So IMHO, they should be there, but whether they should be directly visible to the user as part of an application payload is a different question. Normally this would be transparent to the programer unless they specifically asked for it - and from Damien's reply that was the intent of the function call being used.

pelikhan, 2017-04-21:
Could we move down the specification of the packets used in PXT to the DAL? So that DAL consumers can interoperate on the same C++ struct layouts?

finneyj, 2017-04-21:
Yup - we could also create a new protocol ID for it then it makes interop easier.

I think @DavidWhaleMEF is planning to start a knowledge base article where we could discuss and document supported radio protocols.