lancaster-university/microbit-dal #286: File system is not persistent

DavidWhaleMEF, 2017-04-22:
microbit.org-ticket: 1229

The files can be read out many times BUT a Reset or Power Down will clear the data.

I seem to remember reading that the data file was kept in EEPROM and NOT destroyed by a Reset or Power Down. Can someone please confirm that this statement is true?

DavidWhaleMEF, 2017-04-22:
See also: https://github.com/Microsoft/pxt/issues/1495

finneyj, 2017-04-22:
@DavidWhaleIET 

Correct - all data is held in FLASH memory so is persistent. curious.

Let me try to repro your code in C.

finneyj, 2017-04-22:
Just tested this on native C, which is a pretty close replica of how PXT uses the FileSystem classes, and it seems to work ok. Certainly persists data across a reset and no data corruption that I can see.

Have you got a copy of the PXT code so I can repro?

```
#include "MicroBit.h"
#include "MicroBitFileSystem.h"
#include "MicroBitFile.h"

MicroBit uBit;
MicroBitImage heart("0,255,0,255,0\n255,255,255,255,255\n255,255,255,255,255\n0,255,255,255,0\n0,0,255,0,0\n");

const char *fname = "output.txt";

void initFileSystem()
{
    if (MicroBitFileSystem::defaultFileSystem == NULL)
        new MicroBitFileSystem();
}

void appendLine(const char* filename, const char* text)
{
    initFileSystem();
    ManagedString fn(filename);
    ManagedString t(text);
    MicroBitFile f(fn);
    f.append(t);
    f.append("\r\n");
    f.close();
}

void readToSerial(const char* filename)
{
    initFileSystem();
    ManagedString fn(filename);
    MicroBitFile f(fn);

    while(1)
    {
        int c = f.read();
        if (c <= 0)
            break;

        uBit.serial.printf("%c", (char)c);
    }

    f.close();
}

void onButtonA(MicroBitEvent)
{
    appendLine(fname, "hello");
    uBit.serial.printf("W");
    uBit.display.print("W");
}

void onButtonB(MicroBitEvent)
{
    readToSerial(fname);
    uBit.display.print("H");
}

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    uBit.display.print(heart);

    // If main exits, there may still be other fibers running or registered event handlers etc.
    // Simply release this fiber, which will mean we enter the scheduler. Worse case, we then
    // sit in the idle task forever, in a power efficient sleep.
    release_fiber();
}

```

finneyj, 2017-04-22:
also tried a file system created at a predefined (rather than autodetected) location, as used in PXT. This also seems ok. Test as above but with slightly amended constrcutor:

```
new MicroBitFileSystem(0x2d000,0);
```

DavidWhaleMEF, 2017-04-22:
I've asked the user to send a .hex or a javascript file so we can reproduce this case. I'll let you know when they reply. Perhaps it's a PXT corruption?

finneyj, 2017-04-22:
Thanks @DavidWhaleMEF. I think we'll need that repro case to progress any further.

Hard to say what the cause of this issue is at the moment.

finneyj, 2017-04-22:
this also seems to work ok, and persists data across reset. Generated from pxt.microbit.org/beta

[image]

as does this:

[image]

I'll have to wait for the repro case.

DavidWhaleMEF, 2017-04-22:
repro failure case provided by customer:

The code below corrupts around every 50 numbers. However by playing with the For Loop Count, the corruption changes randomly.

Also for some reason it HAS TO HAVE a DELAY just before it starts writing to the file the first time ( As commented in the code below ). 

I use TeraTerm to view the serial data.

let ay = 0
let file = ""
input.onButtonPressed(Button.A, () => {
    // NEEDS A DELAY HERE!
    //
    // EVEN CORRUPTS THE SMILEY FACE AND THEN CRASHES WITHOUT IT!
    basic.pause(1000)
    basic.showIcon(IconNames.Happy)
    for (let i = 0; i < 250; i++) {
        files.appendNumber(
            file,
            ay
        )
        ay += -1
        files.appendLine(
            file,
            ""
        )
    }
    basic.showIcon(IconNames.Yes)
})
input.onButtonPressed(Button.B, () => {
    file = "data.csv"
    files.readToSerial(file)
})
file = "data.csv"
ay = 250

jamesadevine, 2017-04-22:
@DavidWhaleMEF I just tried this on PXT beta and the exact code you've pasted works fine for me. I got correct output as well from my serial client. I also removed the delay and it continued to work fine.

finneyj, 2017-04-22:
But failed as per description in live pxt.microbit.org though?

jamesadevine, 2017-04-22:
@finneyj Yes

finneyj, 2017-04-22:
ok - thanks @jamesadevine!

@DavidWhaleMEF Sounds to me like this might be an already resolved issue... We know that PXT was mis-calculating the start vector for the FileSystem in version prior to the current beta, and this could have all manner of side effects.

DavidWhaleMEF, 2017-04-23:
Ok thanks, I had thought the customer was testing on beta, but I'll get them to confirm, perhaps they misunderstood instructions. Thanks!

DavidWhaleMEF, 2017-04-23:
Linked PXT ticket has attached .hex file with failing test case. Customer has confirmed that they are using this on pxt.microbit.org/beta

DavidWhaleMEF, 2017-04-23:
https://github.com/Microsoft/pxt/issues/1495#issuecomment-296456566

DavidWhaleMEF, 2017-04-23:
@finneyj Definitely still broken in beta, I just tried this again by rebuilding customers code in beta, and it corrupts data sent to serial

[image]

DavidWhaleMEF, 2017-05-02:
ping - do we know yet if this is a DAL issue or a PXT issue? Ta!

finneyj, 2017-05-03:
@DavidWhaleMEF  I did manage to repro something close to (but not exactly) this on pxt beta. Is most likely on micorbit-dal side IMHO. Prob handling of page boundaries is a little off somehow. Strange this has never surfaced before though.

We'll attempt a native C repro case.

DavidWhaleMEF, 2017-05-15:
ping - is there any progress on diagnosing this issue yet? Is it a DAL issue, or is it a PXT issue, have we narrowed that down yet do you know? Thanks!

finneyj, 2017-05-27:
Managed to reproduce this in C. Fixed in this extremely exciting patch:
https://github.com/lancaster-university/microbit-dal/commit/66d6fb3f81c72c56f7e11e95e5e0999a4da6bda1

Fix now in master, will be in next tagged release (rc10).

closed 2017-05-30 by finneyj