lancaster-university/microbit-dal #289: MicroBitIOPinService truncates analogue input values above 255

martinwork, 2017-05-14:
onDataRead and idleTick both do this: 

uint8_t value;
value = io.pin[i].getAnalogValue(); 

So values above 255 get reported mod 256.

martinwork, 2017-05-24:
@finneyj Have I misunderstood?

finneyj, 2017-05-24:
@martinwork  Apologies for not picking this up sooner. 

This does look odd. I vageuely recall talking with @bluetooth-mdw about using 8 bit values over bluetooth to reduce bandwidth - maybe this is a broken implementation of that? Doesn't look right in any case!

finneyj, 2017-05-24:
Just checked the micro:bit BLE profile:
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

Values are indeed constrained to 8 bit to reduce overhead, but this is clearly not the right 
implementation. Looks to me like there should be s ">>2" on the end that somehow was missed. 

Agree?

cpseager, 2017-05-24:
yes, that would then be consistent with the analog out shift that is already done here:

https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitIOPinService.cpp#L223

finneyj, 2017-05-24:
Indeed. I'll work up a patch.

martinwork, 2017-05-24:
Are there any inputs or outputs for which the edge case 1023 vs 1020 would be noticeable? For reporting an input or setting an output, would it be more natural for an app to use the range 0-1023, as that is microbit's range, rather than 0-255 or 0 to 1020? Is it worth using more complicated formulae than *4 and /4?

finneyj, 2017-05-25:
@martinwork Hmmm... good point.

I think for input, /4 would largely be ok (A to D converters aren't usually *that* accurate anyway, plus 
noise on the line etc). I think the effect of this ~0.3% clipping at the upper end it would be pretty negligible. 

For output it's a bit more of an interesting case though - as the "analogue" output is actually just a PWM. Hence a rounding down of 1023 to 1020 would be the difference between a permanently HIGH signal at ~3V and something that pulses low occasionally. This feels more significant to me, and is worth factoring into the code I feel.

Given this is "rarely used code" (i.e. it's not normally going to be called 100 times a second or anything), I think accuracy would certainly trump processor efficiency in this case. as long as we keep to integer arithmetic (we have no FPU here). 

A linear interpolation formula would probably be ideal:

```
(int) (x * 1023 * 1024) / (255 * 1024))
```

This means a hard zero  => zero a hard 255 => 1023. Everything in between is interpolated linearly. 
Given all those constants, the compiler will take out two of those multiply operations anyway, so it's actually pretty lean. Maximum value in a register would be 255*1023*1024 (~267M), so well within the range of a 32 bit CPU.

For consistency, we should align the input to this formula too (it would be a bit weird if input and output used a different formula). Here I think we want the inverse of the above:

```
(int) (x * 255 * 1024) / (1023 * 1024))
```

We like?

martinwork, 2017-05-25:
Clever stuff - perfect I think.

cpseager, 2017-05-25:
I don't see the point of playing with the analog out formula as it will simply negatively affect existing programs using sounds / servos etc. The 8 bit is quite restrictive for accurate pwm sounds, but tweaking the formula won't help that at all, the only way to resolve that issue is to go to proper 10 bit support. So my vote is to leave it *4 (<< 2) and add in a proper new 10-bit option as well.

For analog in, 8-bit is actually beneficial for most educational projects as it helps filter out some of the noise between successive readings. For 'in' it is not a 0-1020 range, it it still 0-1023 and you already get a 255 at the top end of the 8 bit reading with the proposed simple >> 2.

Most educational projects will be simply looking for a threshold (e.g. light / dark with an LDR) and hence 8 bit (that works correctly!) is fine.

finneyj, 2017-05-25:
Folks, As two of the main users here, I'm happy to go with your agreed preference here.

A three point compromise might be:

1) bug fix the "in" as previously discussed (simple /4). This would be fine and consistent with the current output algo. I think the non linearities are pretty much negligible.

2) *maybe* (I think i'm in favour) we add a special case to out that maps 255 => 1023, but leave everything else unchanged. This just makes the full range accessible at the cost of a minor step at the very top end.

3) Consider a 10 bit BLE API (through 16 bit read/write ops). Getting @bluetooth-mdw's input here would be very insightful.

Thoughts?

martinwork, 2017-05-25:
Since "in" has never worked, what is the problem with starting with the more correct formula?

Your "maybe" sounds good to me.

cpseager, 2017-05-25:
All 3 would be fine for me, thanks. Although I'm getting old, as I just remembered number 3 has already been done and is already available (see #209) !

I think /4 is fine for 'in' - there is no real advantage to using the more complicated formula for input, as in in reality the noise on the ADC would make far more difference to the reading than the slightly more accurate formula would. Successive 10 bit readings can vary by +/- 3 or so, so trying to use a more accurate formula won't compensate for this inherent noise difference anyway..

bluetooth-mdw, 2017-05-25:
"in has never worked"

Indeed not! Issue submitted in 2015!

https://github.com/lancaster-university/microbit-dal/issues/30

finneyj, 2017-05-25:
Folks, please see #292

This tests out cleanly on my oscilloscope , using Nordic's nRF Connect diagnostic tool. Let me know if you're happy with this as a resolution.

closed 2017-05-27 by finneyj

martinwork, 2017-05-28:
@finneyj About onDataRead...

The documentation I can can find doesn't seem very helpful, but the way GattServer calls authorizeRead suggests onDataRead is expected to set the members of the params parameter which causes the new value to be passed to sd_ble_gatts_rw_authorize_reply.
params->data = (uint8_t *)ioPinServiceDataCharacteristicBuffer;
params->len  = pairs * sizeof(IOData);

notify always calls sd_ble_gatts_hvx to send a notification. If the client is registered for notifications two packets arrive: a spurious notification and the reply to the read. If the client doesn't want notifications, sd_ble_gatts_hvx still updates the value and one packet arrives, though the documentation seems not to guarantee the value will updated in this case. Anyway, notify seems to work in both cases. If registered for notifications, the client is unlikely to keep sending reads so the double packets shouldn't be a problem.

However, I think onDataRead could instead use write( handle, ptr, length, true) which simply calls sd_ble_gatts_value_set (because localOnly = true) and doesn't result in an extra packet.

My understanding is that when using notify or write, it is not necessary for the data buffer to exist after the call. Also, when creating the characteristics, the initial value buffer doesn't need to exist after addService returns - in fact is it needed at all when the initial length is zero?

So maybe ioPinServiceDataCharacteristicBuffer doesn't need to be a member variable and could be local to updateBLEInputs.

The design of the read authorisation callback is unfortunate, in that using the modify-params method would require the data buffer to exist after the callback returns.

smartyw, 2017-10-27:
Coincidentally, I'm currently considering using the Event Service to transport 10 bit values from analog reads rather than use the IO pin service.

It's all a distant memory but I think we ended up with design we have as a consequence of trying to allow data to/from multiple pins to be communicated in one operation. But even with that in mind, I think this could be better and we could have our cake and eat it. Right now the pin data characteristic contains an array of one or more uint8 pin number / uint8 value pairs. This is where the loss of resolution comes in. 

But I see no reason why we couldn't instead have an array of one or more 16 bit, composite pin number/value fields, with the least significant 10 bits the value and the 5 bits above that containing the pin number (max 25 decimal, right?). We'd even have one bit to spare!

I know  changing this could impact applications. But I'm wondering if it's worth making the change anyway.

Thoughts?

PR #292 (2017-05-27): Bugfix 289
Fix #289