lancaster-university/microbit-dal #294: Could we control compass calibration over BLE?

martinwork, 2017-05-28:
A suggestion for discussion - no idea yet how this might best be done!

Mainly:

- Abort the calibration routine
- Trigger calibration or re-calibration

And maybe:

- Detect if calibration is running
- Detect if calibration has been done

bluetooth-mdw, 2017-05-30:
I'm actually working on this already. I added a new characteristic but calling the calibrate() function was causing a crash. Discussed with @finneyj  and he thinks it was probably an out of memory problem. Not had time to revisit since but will do so.

[image]

0 - state unknown
1 - calibration requested
2 - calibration completed OK
3 - calibration completed with error

This app uses the magnetometer in two places and was the trigger for me wanting to make this possible over Bluetooth.

http://www.bittysoftware.com/apps/bitty_blue.html

martinwork, 2017-05-30:
Is your DAL code online?

martinwork, 2017-05-31:
I think the crash could be that it is entering the draw-a-circle routine while loop while inside the BLE onDataWritten handler. compass.calibrate creates an event but the handler for this event is registered as "IMMEDIATE", so it's just like calling the code directly.

Not sure if this is the best idea, but onDataWritten could record the request for calibration and have that picked up in an onTick handler. Alternatively, I suppose you could register your own event which isn't IMMEDIATE and call calibrate in that event's handler.

The BLE connection may have to be closed while the calibration takes place. It used to be the case that the calibration data could not be saved while BLE is connected. I don't know if this is still the case. 

A minor point... I think the characteristics could be created with their initial values rather than creating with empty values, the using write to set them. Also the GattCharacterisitc constructor has a parameter "hasVariableLen" which defaults to true. I wonder if it saves memory to set this to false when relevant.

bluetooth-mdw, 2017-05-31:
Thanks @martinwork . This was literally my first, thrown together attempt. On the first test it crashed and I've had no time since. Will see if I can take another look at this soon. Good info regarding the draw circle loop.

smartyw, 2017-06-06:
This works now.

https://github.com/smartyw/microbit-dal/commit/66fd96b098a85936c60f594a4fee8f3afb82090f 

I've not used the hasVariableLen flag or changed the approach to setting initial values. I've stuck with the pattern used throughout the DAL for this. I think these comments / recommendations should go in a new git issue and the whole set of Bluetooth services reviewed and refactored in one pass if it's concluded these changes are beneficial.

smartyw, 2017-07-10:
@finneyj Per our chat a while back, I'm refactored so that all compass related events are now services by the same event handler. See https://github.com/lancaster-university/microbit-dal/pull/296

smartyw, 2017-10-05:
I believe this should be closed as it's been implemented and merged.