lancaster-university/microbit-dal #297: missing check for failed malloc.

drmpf, 2017-06-10:
verify_stack_size does not check for failed malloc

```
void verify_stack_size(Fiber *f)
{
    // Ensure the stack buffer is large enough to hold the stack Reallocate if necessary.
    uint32_t stackDepth;
    uint32_t bufferSize;

    // Calculate the stack depth.
    stackDepth = f->tcb.stack_base - ((uint32_t) __get_MSP());

    // Calculate the size of our allocated stack buffer
    bufferSize = f->stack_top - f->stack_bottom;

    // If we're too small, increase our buffer size.
    if (bufferSize < stackDepth)
    {
        // To ease heap churn, we choose the next largest multple of 32 bytes.
        bufferSize = (stackDepth + 32) & 0xffffffe0;

        // Release the old memory
        if (f->stack_bottom != 0)
            free((void *)f->stack_bottom);

        // Allocate a new one of the appropriate size.
        f->stack_bottom = (uint32_t) malloc(bufferSize);

        // Recalculate where the top of the stack is and we're done.
        f->stack_top = f->stack_bottom + bufferSize;
    }
}
```

jamesadevine, 2017-06-11:
@drmpf

if there's not enough space to page in a new fiber, all hope is lost! :)

I think you could make the argument we do - we override Malloc, which induces a panic when no more memory is available.

James.

drmpf, 2017-06-11:
OK found that code in
https://github.com/lancaster-university/microbit-dal/blob/master/source/core/MicroBitHeapAllocator.cpp

closed 2017-06-11 by drmpf