lancaster-university/microbit-dal #300: Accelerometer service notifications irregular

DasPez, 2017-06-14:
I would like to use the Microbit as a foot mounted sensor. It is connected to an Android device with BLE.
That's where I measure the time between two notifications.
The microbit is configured to `uBit.accelerometer.setPeriod(20);`
The last item is the delta time:
```
-144, 656, 800, 1
-128, 656, 800, 1
-128, 656, 800, 42
-128, 656, 800, 1
-144, 656, 800, 48
-144, 656, 800, 2
-144, 656, 800, 2
-144, 656, 800, 45
-128, 656, 800, 2
-128, 656, 800, 2
-128, 656, 800, 45
-128, 656, 800, 2
-128, 656, 800, 46
-128, 656, 800, 3
-128, 656, 800, 1
-128, 656, 800, 46
-128, 656, 800, 48
-128, 656, 800, 1
-128, 656, 800, 3
-128, 656, 800, 45
-128, 656, 800, 2
-128, 656, 800, 2
-128, 656, 800, 45
-144, 656, 800, 2
```
after changing the period remotely with the Accelerometer Period Characteristic to 2ms:
```
-160, 640, 800, 40
-160, 640, 800, 2
-144, 640, 816, 1
-144, 640, 816, 2
-144, 640, 816, 1
-144, 640, 816, 1
-144, 640, 816, 42
-144, 640, 816, 3
-128, 640, 816, 1
-128, 640, 816, 1
-144, 640, 800, 1
-144, 640, 800, 1
-144, 640, 800, 42
-144, 640, 800, 2
-144, 640, 816, 1
-160, 640, 800, 1
-160, 640, 800, 1
-144, 640, 800, 2
-144, 640, 800, 42
-144, 640, 816, 1
-144, 640, 816, 2
-144, 640, 816, 1
-160, 640, 800, 1
-160, 640, 816, 1
```
20ms period - average: 18ms
2ms period - average: 8ms (LEDs start to flicker)
1ms period - crashes immediately

Should I shedule the sensor updates manually?
Is there a "config.json" way to make the 1ms period work?
I tried changing things there, but most of the times the Bluetooth Service just vanished from my list of available services on the Android device.

kenyoncoder, 2017-06-14:
I also see the micro:bit crash every time I set the notification period to 1ms from my Android App. Longer periods appear to work OK, but I haven't actually checked the timings.

micro:bit flashed with a .hex file from PXT

Android App developed using latest beta release of BLE and microbit extensions in MIT App Inventor.

martinwork, 2017-06-14:
I think the BLE connection can be expected to send a packet of up to 20 bytes every something like 15ms or 30ms, though it will go faster for a time. With the accelerometer service, each reported value uses one packet. So using smaller periods will be too much for the BLE connection and I would expect values will just be dropped. I can't think what would cause a crash.

Maybe you could get better throughput by having the firmware take readings and transmit them in blocks of 20 bytes using the BLE UART service.

finneyj, 2017-06-15:
Folks - thanks for reporting and discussing. 

It looks like we have a few issues here:

1) Setting high frequency updates seems to lock/crash the device. 
2) There's some loss/jitter on high frequency settings.

for 1) we should put a repro together here in Lancaster so we can debug it. @alennie1993 - let's tag team this one. Should be simple to diagnose once we can see it.

for 2) I agree with @martinwork we're probably seeing some loss and buffering in here due to the BLE protocol implementations (either the nordic stack on the micro:bit or the Android stack). 

i do recall this though, which allows a range of BLE packet update times to be configured... 

https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitBLEManager.cpp#L401

I also vaguely recall a discussion with @bluetooth-mdw about the "correct" value for this, and it maybe that the update period is defined by the central device, not the peripheral (i.e. Android). @bluetooth-mdw Any thoughts here on the packet update rate and jitter we might expect over BLE in this case?

cpseager, 2017-06-15:
I found that settings in the central device can make a big difference to responsiveness, When I was using the BLED112 dongle its default power up Connection Event Interval was 75ms. Changing this to 20ms made a huge difference to how responsive the app was to m:b sensor changes. Here's the explanation from @bluetooth-mdw 

_(3) The connection interval defines how frequently the connection may be used by master/slave. A lower connection interval means we get more opportunities to send data. The minimum is 7.5ms. I can see that a value of 75ms is being used. 

So, reducing the connection interval in the connect request from the client (Chrome App) could help since packets will be sent faster and we may avoid running out of TX buffers. Reducing the volume of notifications requested and the frequency with which they’re sent would also help._