lancaster-university/microbit-dal #302: Release rc10 module.json has version 2.0.0-rc9

martinwork, 2017-07-22:
In release rc10 and master, module.json says "version": "2.0.0-rc9".

Note also that lancaster-university/microbit has an rc11 which depends on microbit-dal rc10.

It's getting confusing!

martinwork, 2017-07-22:
@pelikhan 
makecode refers to microbit rc11, so the about box says v2.0.0-rc11, it's using DAL rc10 and the hex files have firmware version 2.0.0-rc9--g

finneyj, 2017-07-22:
Hi @martinwork 

Confusing indeed - thanks for raising here.

I suspect you've just hit an unfortunate one off. @pelikhan kindly tagged a release for me while I was out of town for this version, and I remember hi contacting me at the time saying he a had a minor snafu with version numbers - which is almost certainly why the microbit and microbit-dal repos slipped out of sync here. Normally the microbit-dal, microbit and microbit-samples repos are version locked together.

I'm a little puzzled why the runtime in the HEX is reported rc-9 though. This is generated based on the github tag the binary was built against, so should be accurate. All I can think of is that perhaps the binary MakeCode is using was built before the new tag was applied... 

@pelikhan could that be possible?

martinwork, 2017-07-23:
I also thought it was the tag and assumed that I just didn't understand. I now realise that I was also assuming dal rc10 was being used by make code simply because it was referred to by microbit rc11, so maybe it isn't.

I have just done this test:
- download a zip from microbit-samples: microbit-samples-master
- using yotta on a mac: yt clean then yt build
- flash the hex and check version with light blue
- it's rc9
- change the versions in the module.json in root, microbit and microbit-dal folders (replace the rc9 bit with sam, mic, dal)
- rebuild and flash again
- the version has changed to the one in microbit-dal / module.json

I guess yotta creates YOTTA_MICROBIT_DAL_VERSION_STRING from module.json. So maybe it's not the tag, but the module file in the tagged release.

finneyj, 2017-07-23:
right - looks like the microbit-dal version field in module.json didn't get bumped in the last release... most unfortunate.

pelikhan, 2017-07-23:
We can do a round of "bumps" and link to the new version on the next release