lancaster-university/microbit-dal #305: micro:bit to micro:bit UART

PandaTheOne, 2017-07-28:
hey guys!

First, sorry for my bad English.
 I'm really new in the micro:bit world. I try to experiment UART.
 I have two micro:bit. And I connect them.

my code are :
 ` #include "MicroBit.h"

MicroBit uBit;
 MicroBitSerial serial(MICROBIT_PIN_P1, MICROBIT_PIN_P2);

int main()
 {
 uBit.init();
 uBit.serial.baud(115200);
 int c;
 c=1;
 while (1)
 {
 uBit.serial.send(int(c));
 uBit.sleep(100);
 }
 }
 `

and

` #include "MicroBit.h"

MicroBit uBit;
 MicroBitSerial serial(MICROBIT_PIN_P2, MICROBIT_PIN_P1);

int main()
 {
 uBit.init();
 uBit.serial.baud(115200);
 int c;
 while(1)
 {
 c=uBit.serial.read(ASYNC);
 if(c==1)
 {
 uBit.display.print("A");
 uBit.sleep(100);
 }

}
 } `

So I just want to write "A" on one micro:bit by sending 1 from the other. but It doesn't work haha
 If you have any idea, please help me!

Thank's a lot.

finneyj, 2017-08-07:
Hi @PandaTheOne 

Sorry for not getting back to you sooner - busy week for me last week!

I think I can see what might be going on here. The micro:bit only has support for one serial port, and the Microbit object brings enables this (initially connected to the USB interface) when you create the Microbit object... So most likely, when you try to create that second serial port, it doesn't really attach to anything.

Instead of creating another MicroBitSerial object, try using the ```redirect``` function instead:
https://lancaster-university.github.io/microbit-docs/ubit/serial/#redirect

so something like:

```uBit.serial.redirect(MICROBIT_PIN_P1, MICROBIT_PIN_P2)```

Hope this helps!

PandaTheOne, 2017-08-10:
thanks! I will see what can I do with this one! 👍

microbirdman, 2017-12-17:
Hi guys!

Regarding the UART. I was wondering, is it possible to achieve transfer speeds higher than 115200 bauds. It seems like the Micro:bit is introducing errors when the speed is too high, beyond 115200. I was expecting it to achieve at least 1.5 Mbit when using USB for Tx and Rx. Thanks in advance.

finneyj, 2017-12-19:
hi @microbirdman 

Good question. I haven't performed any direct reliability tests, but have also heard reports of users also seeing errors at higher baud rates.

There could be a number of causes of this:

1) To be honest, the serial port on he nrf51 is not that great. Whilst it can certainly drive faster rates on paper (the data sheet lists up to 1Mbaud for example), the port isn't DMA enabled so at the end of the day we need to rely on the CPU to do byte-by-byte processing. Moreover, the TX side is unbuffered, and the RX side only has a 5 byte deep buffer. That means the flat out, the TX side will be driving a byte every 8uS and the receive side would need to empty its buffer ever 40uS, and within 8uS to prevent data loss. That's quite a big ask (but not impossible) on a 16MHz processor through high level languages.

2) We're just wrapping the ARMmbed serial driver here. Every level of abstraction does bring some cost of course, so we're probably paying a little here.

3) The UART over USB interface also runs through the KL26 DAPLink chip on the board - its possible that this causes some problems at higher baud rates... we'd need to separate out UART tests over an external pin to determine this...

Would you be interested in trying out some tests @microbirdman? and reporting back what you find? I'd be happy to guide.

microbirdman, 2018-01-23:
Hi @finneyj,

Thanks very much for your amazing reply. I am working on a project I call -Bit:boy- that let's you play and create games for the Micro:bit adding an Oled display and so on. My goal is to mirror the Bit:boy's display on my computer's screen via USB. It works but the frame rate is very low as expected with a transfer rate of 115200. I am transferring 1024 bytes via SPI per frame.
I am interested in helping. What can I do? Thanks again.

Interpause, 2018-09-07:
May I inquire if there is any possible way to have more than one serial channel (say, one with the USB and one over the pins) on the Microbit? The documentation gives me the impression that that is possible.