lancaster-university/microbit-dal #306: Feature request: Add option to change the BLE name

microbit-carlos, 2017-08-03:
microbit.org-ticket: 2428

As the BLE advertising name is used to identify micro:bits, it might be useful to be able to set it to something else in code.

Since the name is formed by the `BBC micro:bit [` + friendly_name + `]` string, maybe a MicroBit::setName() method could be added to modify friendly_name or replace the return from MicroBit::getName() to the user-specified string.

bluetooth-mdw, 2017-08-04:
The name is the only way to identify the scanning device as being (probably) a micro:bit. Is there a particular use case where you wish to use a different name?

finneyj, 2017-08-04:
Hi @microbit-carlos @bluetooth-mdw 

Thanks @bluetooth-mdw, this is my concern too. I'm very supportive of being as open and flexible as possible, but we also need to ensure we don't break key user experiences.

@martinwork - As someone looking after a micro:bit BLE app, I assume this might keep you awake at night a bit?

bluetooth-mdw, 2017-08-04:
All my apps look for "BBC" in the name as the only way to deduce an advertising packet is from a micro:bit. Not something I'm hugely concerned about as people by definition, need to be running a suitable hex file on their micro:bit to use the apps. But we do have lots of moving parts already... to pair or not to pair, just works vs passkey, pairing mode available or not available. So there needs to be a good use case to open up an API for this. Obviously you can do it anyway if you really want to and are happy to delve into the DAL and mbed APIs.

finneyj, 2017-08-04:
@bluetooth-mdw - right!

I guess the use cases here would be for cases where the user wants the device to *not* look like a micro:bit... in which case, perhaps breaking compatibility with micro:bit apps is not a problem, provided that expectation is made clear to the programmer.

@microbit-carlos Were there any use cases in the customer ticket that raised this issue?

microbit-carlos, 2017-08-04:
The use case was within a classroom scenario, to be able to easily identify physical micro:bits and their pairing name by labelling them with an easy to understand naming scheme.
For solving this specific use case this way, the customer would need a permanent way to change the name that resists flashing a new program, so that would likely be out of scope. Jonny has already provided the customer with a lookup table to identify the friendly name from the pairing pattern on the screen, and that seems to be a good alternative.

However when I was first thinking about this, I recalled on my own experience having a few micro:bits on my desks or during demos and constantly getting confused about which is which, so it would have been useful to be able to easily identify them on my phone (e.g, having their colour in the name).

So, I wasn't suggesting to change the entire advertising name, as the mobile apps do indeed depend on that. Instead `MicroBit::getName()` returns the friendly name derived from the nRF51 UID, and then the BLE Manager puts that inside the advertising name (`BBC micro:bit [` + MicroBit::getName() + `]` ). So if we provide a `MicroBit::setName()` method that only affects the return value from `MicroBit::getName()`, then you would still have a valid  `BBC micro:bit [myPersonalName]` string.

Do we have any hard constrains on the length of the friendly name?

finneyj, 2017-08-04:
thanks @microbit-carlos 

There are tight constraints on the friendly name at the moment - there's a direct mapping between that and the the pairing-code histogram... So if we allow freetext names, then identifying a micro:bit from a histogram becomes impossible (which again will have an impact on micro:bit companion apps that try to pair/program a micro:bit...)

martinwork, 2017-08-05:
I am 100% in agreement with @bluetooth-mdw 's point about "moving parts"! And changing the advertised name of a microbit paired with iOS could be the stuff of nightmares! We wouldn't want every hex to change the name.

I believe it's by design that the user specifies the microbit for pairing by copying the pattern from the LEDs, rather than choosing from a list of advertising devices. Changing the [xxxxx] part would require significant changes to the app UI and probably mean scrolling xxxxx on the LEDs in pairing mode.

It would cause less upheaval to modify apps to accept variations in the "BBC micro:bit " part and to display the custom name once paired. Losing "BBC micro:bit" could be awkward for documentation and support. Perhaps a custom suffix: "BBC micro:bit [zotup]Yellow03"? Is the max length 29? Could we shorten "BBC micro:bit "?

microbit-mark, 2017-11-09:
Adding a comment here, as I have another request for this in support, again as a useful feature for classroom teaching.

bluetooth-mdw, 2017-11-09:
In what respect would this be useful for classroom teaching? Could you elaborate please @microbit-mark ?

microbit-mark, 2017-11-09:
As with comment above from @microbit-carlos, the teacher has asked for an easy to understand naming/numbering scheme to identify micro:bits in the classroom.

cpseager, 2017-11-09:
@bluetooth-mdw I can see the benefit of a nickname but I would vote for leaving the 0x09 name data alone in the advert and create a new 0xFF configurable 'nickname' instead. That won't break or change anything that currently exists, but gives the app designers in the future the option to show the real name and/or the user configurable nickname as they choose.

microbit-carlos, 2017-11-09:
Just to add a quick note, in think the case @microbit-mark is mentioning is when the user is  looking at a situation when the micro:bit app is not used to pair, and goes through the normal mobile OS pairing process instead.

pelikhan, 2018-11-03:
> It came to my attention that some of the teachers have extremely hard-time teaching using micro:bit in the classroom while there are no notebook is available. They rely on galaxy tablets to use app to develop and download. However, when trying to connect using bluetooth, all of micro:bit has same ID, and therefore around 20 children has to take turn to download. Just by doing so, half of the classroom hours get passed by just for spending on download 1 update. They even think about change to arduino with bluetooth module to replace micro:bit for that matter if this matter is not solvable.

Have you tried the “no painting mode” in MakeCode, it allows to connect without paring — thus the micro:bit keeps the 4 letter name in the device name display.

henrykim1004, 2018-11-03:
Can I change 4 letter name in the device name on the device?

pelikhan, 2018-11-03:
Microbit is an edu device and the classroom scenario is/should be the main scenario here. The current naming scheme does not cut it and does not work with WebBLE either.

pelikhan, 2018-11-03:
Bluetooth is not limited to apps anymore, it can be used from Windows, Linux, MacOS, Android through the Chrome browser + WebBLE. We should maybe revisit the pairing experience and optimize for school usage.

henrykim1004, 2018-11-04:
I know everything you have said is true for small personal scale or for small group.  However, this is the situation that sound very odd but in fact so common in South Korea. (I am volunteering to help them to solve this matter for Korean Teachers now.)   In this nation, school are quite big; each class has about 30 pupils,  there are about 5 classes per grade, 1st to 6th grades in a elementary school, which sums up as quite a big population per school, and there is ONLY ONE COMPUTER LAB.   Therefore, alternative method for easy level of STEM education is usually done in classroom.   Also due to high number of students, wifi is not available other than computer lab.  (well, most of DHCP ip were already taken by smartphone already).     That is why we need to use galaxy tablet WITHOUT internet, and download code to micro:bit using bluetooth.   Using OTG dongle is not an option and banned since children can easily damage tablet's connection port.   That is why it is extremely important to set each micro:bit to have different bluetooth NAME.

bitsboxscott, 2020-08-04:
Hey everyone, I'm facing a similar desire, wanting to add some customization to the name for Web Bluetooth. Specifically to handle the case where there are multiple kids (in a classroom or siblings at home) coding against Micro:bits from the web. Thanks!

TurquioseTNT, 2020-08-15:
Can we not just edit the update files to update the name to what we want it to be?
here is my code if you want to try it :)
[microbit-Smart-Watch-MicroBit.hex.zip](https://github.com/lancaster-university/microbit-dal/files/5078435/microbit-Smart-Watch-MicroBit.hex.zip)

richgriswold, 2023-06-19:
If the name cannot be changed, then I would suggest tweaking the encoding table.  My daughter's micro:bit has the unfortunate name of "tetuz", which sounds almost exactly like "tetas", or "tits" in Spanish.  My daughter is bilingual, and she picked up on that immediately.  Since the micro:bit is being used in schools with bilingual students, it isn't very professional to have your device called "tits".  Changing the set of consonants for the first or third value would avoid this.