lancaster-university/microbit-dal #308: Feature request: Timer interrupt for Javascript Blocks Editor (powered by MakeCode)

rharbird, 2017-08-10:
Hello,

We would like to be able to put the Micro:bit into a deep sleep and periodically wake it to do something. This ticket is about being able to create a timer interrupt to perform some action periodically using the Javascript Blocks Editor.

Use case: We teach 'A' level students and undergraduates using the Micro:bit. We ask them to code the Micro:bit for IoT projects in which they must gather data over time and it is important to be able to conserve energy on the device. These students are not Computing 'majors' but do need to learn the importance of coding and engineering through working on real world projects. In a recent project students had to create a system for water and presence monitoring in the toilets to ascertain the effectiveness of water saving measures. 

Regards,

Rae

finneyj, 2017-08-10:
Hi @rharbird 

That sounds like an amazing project - brilliant! Exactly the sort of thing we want to encourage for micro:bit.

So, a couple of things here. For a periodic callbacks, you *could* just write  a simple forever loop with a BIG delay in it (the "pause" block in MakeCode). This will actually work pretty well, as MakeCode runs on top of the microbit-dal scheduler, so that delay won't prevent other code blocks from executing, and whenever there's nothing to do, the scheduler puts the CPU into a sleep state.

I do have some code for periodic callbacks that I wrote for a different platform that I'll look to integrate though. 

How much power efficiency do you think you need? If you do the above and disable the display and radio, I expect you'll get down to about 4mW...

p.s. for my sins (which are both great and numerous), I also head up the CAS region centre for the NW UK here in Lancaster, so i'd be very interested to hear your feedback on how you think this project goes.

rharbird, 2017-08-12:
Hello @finneyj ,

Yes, the students get a lot out of these projects. We would really appreciate it if periodic callbacks were available.

I spoke to my colleague, Prof. Steve Hailes, and he says that it would be good to be able to reduce the power usage to mA, rather than mW in deep sleep. In this way, you could decide what you're going to measure and leave your kit in location for about a week, say, without running out of battery. In the case of the water usage monitoring, above, the device could be put into a deep sleep state overnight - we know no-one uses the toilets between 10pm and 6am.

re: P.S: We've run this scenario for 3 years now, we've used a variety of boards. This year, we used the micro:bit with the Javascript Blocks Editor (powered by MakeCode).  I think more students tried, and enjoyed, the coding. I'd be  very happy to have a chat with you about it. 

Rae

DrSianArmstrong, 2018-09-05:
I want to have a timer interrupt using javascript blocks too. Anybody out there know if it has been done? I know it is theoretically possible by programming in C/C++ but I need to retain the ability to build in blocks. I've posted on several forums and asked via help ticket but getting nowhere.

pelikhan, 2018-09-05:
@DrSianArmstrong  You mean someting like ``setTimeout`` in JavaScript?