lancaster-university/microbit-dal #310: Feature request: Ability to include C++ with Javascript wrapper for Javascript Blocks Editor (powered by MakeCode)

rharbird, 2017-08-12:
Hello,

We would like to be able to write packages that contain C++ code with a javascript wrapper. 

The first things that we would like to do with this functionality are:

1. To include the SdFat library,
2. To include a lightweight networking protocol.

We know that we can use the Adafruit data logger for no. 1 ( thanks @DavidWhaleIET ) but in the longer term, we would prefer the lower power option of an sdcard writer.

Regards,

Rae

DavidWhaleMEF, 2017-08-12:
It is already possible to bundle C++ libraries with MakeCode.

This particular package on github is written in assembly language: https://github.com/Microsoft/pxt-neopixel

You'll need the CLI tools in order to build the C++ I think...

https://makecode.com/cli

It's not well documented how you do this, but if you ask @pelikhan I am sure he will point you in the right direction.

Users can then just AddPackage and the cpp (and/or asm) comes in with it, and they can use it without building code. It's how many of the standard MakeCode packages already work I think.

DavidWhaleMEF, 2017-08-12:
Regarding a lightweight network protocol, Rob May from Cisco was implementing MQTT/SN on micro:bit a while ago. https://twitter.com/TigerRob

Also, we have been working really hard with one of our content partners, who have developed a complete course on networking. This course includes a range of unacknowledged, acknowledged, error prone and error free network mode packages for MakeCode, both wired and wireless. I *think* the current plans are that these will surface later this year.

rharbird, 2017-08-14:
Thanks for the update @DavidWhaleMEF. 

We will try the AddPackage feature and are awaiting the new networking functionality!

Rae

closed 2017-08-14 by rharbird