lancaster-university/microbit-dal #311: Eddystone beacon

ckuehnel, 2017-08-15:
I tried to compile the example from showio/microbit-physicalweb and the lancaster-university/microbit-samples/source/examples/bluetooth-eddystone-url/ but in both cases I got errors like 

[image]

What's wrong on my side?

bluetooth-mdw, 2017-08-15:
Works fine using Yotta. So the issue is presumably mbed IDE related. 

[image]

Have you enabled eddystone_url support in MicroBitConfig.h?

```
// Enable/Disable availability of Eddystone URL APIs
// Set '1' to enable.
#ifndef MICROBIT_BLE_EDDYSTONE_URL
#define MICROBIT_BLE_EDDYSTONE_URL               1
#endif
```

ckuehnel, 2017-08-15:
I added the lines into MicroBitConfig.h but got the same errors.
I will try Yotta too.
Thanks,
Claus

Dr. Claus Kühnel - Talstr. 39a - CH-8852 Altendorf

[image] [https://user-images.githubusercontent.com/8341666/29320197-5df48dc8-81ce-11e7-9102-fec3b3323f12.png]

bluetooth-mdw, 2017-08-15:
It looks like the DAL in mbed is behind DAL as used from github and it simply doesn't have the MicroBitEddystone.h class

mbed:
[image]

offline DAL source files:

[image]

So you need to use Yotta and we need to ask @finneyj and @jaustin when mbed will be updated :-)

ckuehnel, 2017-08-15:
An update of the mbed environment is my preferred way because I use other microcontrollers in the mbed IDE already and could stay there. Hope that @finneyj [https://github.com/finneyj] and/or @jaustin [https://github.com/jaustin] have good news.

-------------------------------------------------------------------------
Dr. Claus Kühnel - Talstr. 39a - CH-8852 Altendorf
-------------------------------------------------------------------------

[image] [https://user-images.githubusercontent.com/8341666/29324080-d1854a0a-81da-11e7-94dc-c9089e1bc3ea.png]
[image] [https://user-images.githubusercontent.com/8341666/29324426-e1c450c2-81db-11e7-9104-694fa78f13a3.png]

finneyj, 2017-09-11:
Hi @ckuehnel @bluetooth-mdw 

Thanks for the detailed report. The online mbed build is indeed quite a long way behind the releases we have here in github. I feel your pain @ckuehnel.

The issue is that it's actually pretty difficult to key the online mbed codebase sync'd, as the last time we looked there was no way to sync with mbed.org with a github repo. in addition we've also made several customizations to some of the mbed implementations for microbit that weren't always possible to merge back into the master mbed source tree.

Having said that, we do need a solution to this, and mbed.org is now woefully out of date.

@jaustin Is there someone at ARM who might be able to advise/help out?

p.s. Apologies for slow response - I've been away on vacation until recently.