lancaster-university/microbit-dal #313: Suspected non-randomness of RNG

DavidWhaleMEF, 2017-08-19:
Spotted this on twitter... @finneyj care to comment? it looks quite 'periodic' to the user:

https://twitter.com/alynnafoxie/status/898597401132400641

DavidWhaleMEF, 2017-08-19:
From customer:

Micropython. I started getting better results from getrandbits() than randint().  Makes me wonder if their implementations are different..

DavidWhaleMEF, 2017-08-19:
MicroPython implementation is here:

https://github.com/bbcmicrobit/micropython/blob/a11ce23fd20c6bdc8f33e3c1cf7a268c00cc6281/source/microbit/modrandom.cpp

DavidWhaleMEF, 2017-08-19:
MP uses the dal RND under the bonnet....

#define randbelow(n) (MicroBit_random(n))

finneyj, 2017-09-11:
@DavidWhaleIET We are using a very lightweight "just about good enough" random generator - a LFSR. This was really just to keep codesize down, very early in the project when we were highly concerned about memory footprint. It's certainly not appropriate for any crypto use etc.

LFSR generators are somewhat periodic, but they are simple to compute - documented here with thinks for those interested:https://github.com/lancaster-university/microbit-dal/blob/5746eb94a8fbe944f474c89c3edd237731cb0f0a/source/core/MicroBitDevice.cpp#L270

It shouldn't be *that* periodic though (32 bit cycle), and seeded by a crypto grade generator... A little surprised if it's noticeable by typical end users. If it becomes an issue we could consider just using the libc rand() implementation which would give more entropy.