lancaster-university/microbit-dal #314: problem with Module.json file

DebzM, 2017-08-20:
Found error in build

(workspace) C:\Users\seand\projects\tutorial>yotta init
Enter the module name: <tutorial> example for youtube
Enter the initial version: <0.0.0> 0.0.0
Is this an executable (instead of a re-usable library module)? <no> yes
Short description: example
Author: Sean Dunlevy
What is the license for this project (Apache-2.0, ISC, MIT etc.)?  <Apache-2.0> free

(workspace) C:\Users\seand\projects\tutorial>yotta target bbc-microbit-classic-gcc
warning: tutorial has invalid module.json:
warning:   name value u'example for youtube' does not match u'^[a-z]+[a-z0-9-]*$'
info: get versions for bbc-microbit-classic-gcc
info: download bbc-microbit-classic-gcc@0.2.3 from the public module registry
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.3 from the public module registry

jamesadevine, 2017-08-20:
Hiya,

You need to enter the module name with dashes rather than spaces.

If you haven't already, take a look at the microbit-samples project:

https://github.com/lancaster-university/microbit-samples

James.

Enter the module name: example for youtube
Is this an executable (instead of a re-usable library module)? yes

DebzM, 2017-08-21:
thanks james, will give it a go, and yes i am using the tutorial you suggested - was hoping it was something simple i had missed, will let you know