lancaster-university/microbit-dal #316: BLE adc value change notifications lost

cpseager, 2017-08-23:
I've been testing #292 with a 10k pot on P2 as a potential divider across the power rails (m:b is powered by USB)

Turning the knob slowly works well, the range shown in the app from the adc notifications received is 0-248 (not sure why not 255, but that doesn't really matter!).

However turn the knob rapidly and it is clear that some of the value change notifications are lost. This looks very bizarre as you can change the pot quickly back to the 0V end stop and still be left with a random value, say 28, on screen instead of 0. Nudge the pot very slightly and it will then go to 1 or 2 as expected.

Is there any way around this?

finneyj, 2017-09-11:
Thanks @cpseager. 

Hmm, that definitely doesn't sound like the ideal behaviour. :)

I just reviewed the code, and I'm pretty sure that the only way that you would see this behaviour is if the BLE notify operation is dropped in or after the notify() call to the Nordic SoftDevice BLE stack. This gives us two likely fail cases:

1) The fast(er) rate of notify operations caused by moving the pot quickly results in a buffer overflow of some sort in the BLE stack, and its using a drop tail queue strategy (so the most recent updates are lost).
2) The data is being lost "on the wire" due to the increased data rate.

1) Sounds most likely to me. Although @smartyw might have some ideas of likely causes too.

@cpseager As you have the test case, can you see if the call to notify() here is returning any error codes etc? https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitIOPinService.cpp#L158

It looks like it might report if there are any internal errors in the stack:
https://github.com/lancaster-university/BLE_API/blob/master/ble/GattServer.h#L205

p.s. Sorry for leaving this hanging - been camping in a field in France for a while. :)

martinwork, 2017-09-11:
nRF5xGattServer::notify() returns either BLE_ERROR_NONE or BLE_STACK_BUSY for any error.

notify() calls sd_ble_gatts_hvx. The comments above sd_ble_gatts_hvx say it will "consume an application buffer" and refers to sd_ble_tx_buffer_count_get to find out how many buffers there are (6?).

Once all buffers are used, notify will update the attribute value but fail to send the PDU and return BLE_STACK_BUSY.

I think to guarantee that the latest value gets reported you have to check for BLE_STACK_BUSY and retry, perhaps using an onDataSent callback.

martinwork, 2017-09-11:
A related potential problem is that if the input values are changing all the time, MicroBitIOPinService could swamp the application buffers and block out other services.

martinwork, 2017-09-11:
Retry could be added to MicroBitIOPinService by only updating ioPinServiceIOData if notify succeeds.

finneyj, 2017-09-12:
Indeed - thanks @martinwork. That's what I was thinking too! 

If @cpseager can confirm this is what's happening here, then we can handle this pretty easily using that approach.

bluetooth-mdw, 2017-09-12:
I'm pretty sure we've been here before and there's another issue in github somewhere about this. The problem we faced at the time was that whilst the Nordic APIs returned the responses that @martinwork mentions, the mbed API just swallowed them. Maybe this has changed and the mbed APIs now return these response codes?

cpseager, 2017-09-12:
Sorry, my current dev kit setup consists of emailing @bluetooth-mdw and waiting for a hex file to magically appear, so I can't do any error code debugging at present! However I agree that 1) seems to be what is happening.

What is important for us is that the 'latest' reading is sent, if an intermediate one is lost due to no buffer space that doesn't really matter. So the 'retry' would only need to send the latest reading, not retry to send _every_ missed reading.

martinwork, 2017-09-12:
This is related to #300 

I mentioned about using the return value of notify in #259 , where there is also mention of swallowed responses!

I don't think notify has changed. The only thing is that it reports all errors as BUSY - probably not an issue as anything except no available buffers is probably pretty terminal!