lancaster-university/microbit-dal #318: "Pairing mode" could become a more generic 'utility mode'

jaustin, 2017-09-28:
(this issues should theoretically be filed against microbit not microbit-dal but I think this is a better location for eyes/discussion - notice that this would still likely be *implemented* in the microbit class so that people integrating the DAL would be able to optionally take the behaviour)

The current 'pairing mode' is already mis-named - it's used for BOTH pairing and also updating the hex file when the micro:bit is running a program with the BLE radio.

However there are other things people might like to do that are 'under the hood' to their program
* Measure and display the battery charge level (https://os.mbed.com/users/MarceloSalazar/notebook/measuring-battery-voltage-with-nordic-nrf51x/ )
* Send files from the filesystem over USB serial
* Erase compass calibration

More dramatically, if flash allowed we could even include the BLE services and aserial messagebus bridge so that a more complete 'remote control mode' could be achieved without having to re-install a program.

@finneyj and @jamesadevine I know this has long been in your minds, and @martinwork and @bluetooth-mdw the idea of clarifying that 'PAIRING MODE' is *not* only about pairing is important.

jaustin, 2017-09-28:
Note for battery level, the reading is quite simple - here's an example: https://os.mbed.com/teams/microbit/code/microbit-battery-test/

drmpf, 2017-09-28:
Can we change the 'standard' microbit ADC to use this band gap ref as well.  Currently the ADC output drops as the battery volts drop.

jaustin, 2017-10-31:
@drmpf changing what happens by default is an issue because it'd be a breaking change for a bunch of accessories, etc.

Consider the fairly common case of a slide-switch potentiometer wired off the edge connector - using ADC readings relative to VDD is more useful in that situation...

I think if we were to do the following
* Expose the ability to set the ADC reference voltage
* Expose the ability to set the ADC's input to the supply as well as the pins

then we'd be able to let people write more useful scripts. @dpgeorge may have a different ADC implementation in MicroPython, and I think we should do similar there. 

I've spawned #322 to track the ADC flexibility side and we can continue the 'utility mode' discussion here.

jaustin, 2017-11-07:
Also, when the micro:bit battery is low, flashing over BLE can be inconsistent or pairing won't work. We could set a battery threshold below which we abort pairing mode and instead say "battery too low"?

martinwork, 2018-08-09:
Low battery level can cause BLE flashing to fail halfway, leaving micro:bit soft bricked. It would be good to have the ability for the client app to check battery level before flashing.