lancaster-university/microbit-dal #320: Are we able to control the amplitude (volume) of the sound generated on a pin

microsoftsam, 2017-10-26:
In pxt-microbit, we're using analogPitch to generate sound on a pin. 
Given that we are generating the sound using analog write rather than digital, is it possible to control the amplitude of the sound generated. 
My first inclination was to alter the HIGH analog value sent to the dal. I tried changing 512 to 10, but it seemed that the frequency was affected but not the volume emitted. 

Any thoughts @finneyj?

```
 void analogPitch(int frequency, int ms) {
      if (pitchPin == NULL)
        analogSetPitchPin(AnalogPin::P0);
      if (frequency <= 0) {
        pitchPin->setAnalogValue(0);
      } else {
        pitchPin->setAnalogValue(512);
        pitchPin->setAnalogPeriodUs(1000000/frequency);
      }

      if (ms > 0) {
          fiber_sleep(ms);
          pitchPin->setAnalogValue(0);
          // TODO why do we use wait_ms() here? it's a busy wait I think
          wait_ms(5);
      }
    }
```

cpseager, 2017-10-26:
Adding any potentiometer (around 10k to 50k) between m:b and the jack plug will do the job.

[image]

microbit-mark, 2018-09-06:
Polling this ticket as we've had another feature request for this in micro:bit-support 17464

DavidWhaleMEF, 2018-09-06:
@samelhusseini I think you may be misunderstanding how the sound generation on micro:bit works, and specifically write_analog.

write_analog and set_pitch both generate rail-to-rail PWM. 

For write_analog, the default PWM period default is quite high, so the effect of writing a low analog value is that the PWM duty cycle is mostly off (which gives an average voltage that is very low). Setting a high analog value generates a higher PWM duty cycle, so the high-time lasts longer and the average voltage is closer to 3V.

But the instantaneous voltage of PWM is either 0V or 3V - there is no real in-between.

So, it's not really writing an analog value, it's setting the PWM duty cycle somewhere between 0% (always 0V) and 100% (always 3V) with a proportional fit from 0..1023 mapping to 0%..100% duty cycle. So, 512 would set a 50% PWM duty cycle, which is 0V for half the period and 3V for the other half of the period.

To control volume, you'll need some external circuitry, as suggested above.

It would be possible to use an external digipot, which could then be controllable by the micro:bit - e.g. 

http://www.microchip.com/design-centers/analog/data-converter/digital-potentiometer

Hope this helps.

finneyj, 2018-09-06:
It might be interesting to try altering the duty cycle of the PWM though. We're currently running at a 50% duty cycle. Moving to a lower (or higher) value would mostly affect the tone mostly I suspect, but might also have a perceptible impact on volume...