lancaster-university/microbit-dal #323: Use custom version of atoi() to eliminate dependency on newlib's locale and save RAM

dpgeorge, 2017-11-15:
As discussed over at the MicroPython port to the micro:bit https://github.com/bbcmicrobit/micropython/issues/363, the latest version of newlib (2.5.0) introduced some new locale handling code and has a large (~350 bytes) `__global_locale` data structure that is not constant and lives in RAM.  On the micro:bit 350 bytes of RAM is a lot and it would be good to reclaim this.

Upstream at newlib they say that the only way around this is to add an option to newlib to completely disable the locale stuff, see https://sourceware.org/ml/newlib/2017/msg01041.html

Since that option might not be feasible for us (it requires a custom newlib build) I found another way: the only place in the DAL that uses the locale-dependent code (at least in the MicroPython port) is in `MicroBitImage::MicroBitImage(const char *s)` when it calls `atoi()`.  By simply replacing `atoi()` with a call to a custom version of this function (eg `my_atoi()`) that has a very simple implementation, one can reclaim 364 bytes of RAM and around 600 bytes of flash/ROM.

Is this a modification (changing atoi to my_atoi) that would be accepted in the DAL?

finneyj, 2017-11-15:
Hi @dpgeorge 

Yes, of course. No problem at all - a natural home would be here: 
https://github.com/lancaster-university/microbit-dal/blob/master/source/core/MicroBitCompat.cpp

Do you have a simple implementation to hand? If not, I have a few hundred minions learning C at the moment that could sharpen their teeth on this... They'll do anything for a mars bar.

dpgeorge, 2017-11-15:
Please see above PR for an idea to remove atoi().  Since the code calling atoi() was almost parsing the number already, I just adjusted it so it did construct the integer value.

finneyj, 2017-12-19:
yup - now merged, so closing this off. Thanks @dpgeorge.

closed 2017-12-19 by finneyj