lancaster-university/microbit-dal #327: Can't use PWM Servo & Melody concurrently

microbit-mark, 2017-12-07:
Reposting here as this was referred back as a DAL issue

https://github.com/Microsoft/pxt/issues/3497

This program demonstrates playing a melody and once complete the servo moves 90. waits, then returns back to 180 and wait before looping round.

When inserting the two blocks together there is an interference on the micro:bit device.
Having the melody block on its own is fine. Have the two servo blocks on its own works fine.

[image]

I have confirmed with @finneyj, that PWM shares a single timebase, so servo and audio cannot be used simultaneously. This means the behaviour in the simulator does not reflect the hardware.

Is it possible to address this in the DAL?

microbit-mark, 2018-01-10:
Hey @finneyj just nudging this one for an update

jaustin, 2018-02-15:
After discussion on tech call today, we want to investigate throwing a runtime error for resource availability when users try to access concurrently the same PWM resource with different configuration.

microbit-mark, 2018-10-02:
support-microbit:17850
Had another ticket raised about this behaviour, so polling here for an update.

martinwork, 2018-10-02:
To make this work, sort of, ensure the pins are not trying to use PWM at the same time...
- insert a pause to allow the melody to finish before moving the servo
- insert "digital write pin P1 to 0" at the end of the sequence
But this will mean the servo is not held in position while the tune plays.

CaptainCredible, 2020-09-18:
It would be huge to be able to use more of the hardware timers and have more than one tone at a time (from different pins) or like i'm trying to do, move 2 steppermotors at different speeds simultaniously

whaleygeek, 2020-09-18:
@martinwork FYI I did some experiments with 4 servos, and due to dynamic allocation, the last used servo 'sneezes' back to 0 temporarily as it is no longer receiving a pulse train. This is related to how a single GPTE block on the CPU is used to generate a common timebase and only 3 channels can use it simultaneously.

whaleygeek, 2020-09-18:
@CaptainCredible Hi Captain, loving watching your plotter build. As I understand it, the GPTE (gpio timer block) is used to generate the timebase for PWM for the whole block, which is fine if you are driving 3 servos as they have the same timebase, but sound clearly varies the PWM timebase to simulate a frequency change, which is why you don't get things cooperating here. 

It's always possible to generate toggles in a software way with an extra timer interrupt handler, but this puts significant load on the CPU and can affect latency on other services (I had same issue with an Arduino recently and moved an IR sender carrier timebase off of a standard pin and onto a OCA pin that can generate rock solid timing in hardware with no software involvement, and all the jitter went away, and CPU load went down).

I'd have to refer to @finneyj as to whether there is a separate hardware timer that can still toggle pin state independently of the GPTE block, or whether the (slower) requirement of a 20ms timebase for PWM could be safely scheduled in software.

I must admit, my workaround here was always to use two micro:bits, and either signal between them with radio messages, or send uart messages between them over a single wire and gnd.

FYI, my touch sensing guitar gets 18 touch inputs this way by using two micro:bits connected with a unidirectional UART link: code here: https://github.com/whaleygeek/microbit-midi-guitar/tree/master/src video here: https://www.youtube.com/watch?v=gpyBcDmcHtc

To get two simultaneous different sound playback (melody and harmony) I used the radio approach, it worked pretty well. Video and code here: https://www.youtube.com/watch?v=GqXHkBEngNQ

whaleygeek, 2020-09-18:
@CaptainCredible P.S. Is it not already possible to move two stepper motors at two different speeds simultaneously?

The speed of the stepper is dictated (for positional) from the rate of change of the PWM value. For continuous rotation servos, the speed of rotation is dictated by the actual PWM value.

In both cases, as long as the servos both use the same (traditionally 20ms) frame timebase, your application can generate the change in PWM values independently for the two channels at any rate it wants to?

whaleygeek, 2020-09-18:
@CaptainCredible Pseudo Code for what I mean (for positional servos)

```
onstart
  valA = 0
  valB = 0
  dirnA = 1
  dirnB = 1
  stepA = 10 // note different from B
  stepB = 20 // note different to A
end

forever //servo A
   servoA.setValue(valA)
   valA = valA + dirnA * stepA
   if valA > 180 or valA < 0:
       dirnA = -dirnA
   pause(10)
end

forever //servo B
   servoB.setValue(valB)
   valB = valB + dirnB * stepB
   if valB > 180 or valB < 0:
       dirnB = -dirnB
   pause(10)
end
```

martinwork, 2020-09-20:
@whaleygeek With 4 servos how do you take control away from one of the first 3? Maybe best to do it between pulses, rather than cutting a pulse short? I didn't think stopping the stream of pulses would move a position servo, just release its hold on it's position.

https://makecode.microbit.org/_0VhW6RJDWWa1 sort of does what the OP asked with an occasional stutter.

https://makecode.microbit.org/_3oP4CR98wCuJ waves a flag while playing and displaying happy birthday, and stutters occasionally to show why PWM is best!

I guess the stutters are due to interrupts affecting the timing. I'm surprised in both cases that it isn't worse. I don't know what tolerance a servo applies to individual pulses before it feels it needs to move. I wonder if an mbed Timer could control the pulses well enough to replace PWM, as an alternative for >3 servos or with music.