lancaster-university/microbit-dal #337: Only one service or attribute is registered.

bluexmas, 2018-01-24:
BLE addService 
Only one service or characteristic is registered.

bluetooth-mdw, 2018-01-24:
@bluexmas I think we need a but more information to diagnose this issue. Could you provide some code please, including your config.json file? If your code is correct, the likelihood is you haven't reserved enough memory for the GATT attribute table using the gatt_table_size attribute like this:

```
{
    "microbit-dal": {
        "bluetooth": {
            "enabled": 1,
            "pairing_mode": 1,
            "private_addressing": 0,
            "open": 1,
            "security_level": "SECURITY_MODE_ENCRYPTION_NO_MITM",
            "whitelist": 1,
            "advertising_timeout": 0,
            "tx_power": 7,
            "dfu_service": 0,
            "event_service": 1,
            "device_info_service": 1
        },
        "gatt_table_size": "0x600"
    }
}
```

bluexmas, 2018-01-24:
thank you.
I solved this problem..

closed 2018-01-24 by bluexmas

bluetooth-mdw, 2018-01-24:
Could you share your problem and solution so that it might help others, please?

bluexmas, 2018-01-24:
source - https://github.com/lancaster-university/microbit-samples/blob/master/source/examples/bluetooth-services/main.cpp

config.json - Only one service or characteristic is registered.

{
    "microbit-dal":{
        "bluetooth":{
            "open": 1
        },
        "debug":0
    }
}

config.json - success
{
    "microbit-dal": {
        "bluetooth": {
            "enabled": 1,
            "pairing_mode": 1,
            "private_addressing": 0,
            "open": 1,
            "security_level": "SECURITY_MODE_ENCRYPTION_NO_MITM",
            "whitelist": 1,
            "advertising_timeout": 0,
            "tx_power": 7,
            "dfu_service": 0,
            "event_service": 1,
            "device_info_service": 1
        },
        "gatt_table_size": "0x600"
    }
}